openhwfoundation/cva6 #2033: Parametrization: dependencies with HPDCache

JeanRochCoulon, 2024-04-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

To finish the parametrization task, we would be pleased to remove the declaration of the following variables from 65x_config_pkg.sv. But these localparams are used by hpdcache which prevent from completing the task.

[image]

cfuguet, 2024-04-12:
The noose is tightening ! ;)

I will work ASAP on this.

cfuguet, 2024-04-12:
By the way, @MarioOpenHWGroup were you able to add a HPDcache configuration in the new CI infrastructure based on cva6.py ?

cfuguet, 2024-04-16:
A quick update on this, I have a first version of the HPDcache using module parameters (instead of the package).  It works in our internal standalone testbench with Verilator. I will now try to simulate with other tools and synthesis to be sure that the modifications are well supported.

After that, I will push :)

JeanRochCoulon, 2024-04-16:
Nice !

JeanRochCoulon, 2024-05-19:
Fixed by #2115.  Thanks to @cfuguet.

closed 2024-05-19 by JeanRochCoulon