openhwfoundation/cva6 #2046: How to flush the data cache?

Juan-Gg, 2024-04-16:
I need to flush the data cache to read from memory that has been modified by an accelerator.

I suppose it can be done with a CSR instruction, yet the [documentation](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/CSR_Cache_Control.html) is not there yet.

I tried the fence instruction but it did not work.

I have been looking at the System Verilog code for the caches, there is a `dcache_flush_ctrl_cache` signal driven by the `controller` module, but it seems that it will only work with a Write Back configuration, while my current one is Write Through.

Any ideas?
    Juan

Juan-Gg, 2024-04-16:
Issue #410 notes the same problem, and solves it by enabling and disabling the data cache, by writing to the CSR_DCACHE register.

Looking at riscv_pkg.sv, this register is now in 12'h7C1, not in 12'h701 as in the mentioned issue. I have found the following to work for me.

```
#define FLUSH_D_CACHE() ({__asm__ volatile("csrwi 0x7C1, 0x00"); \
						  __asm__ volatile("csrwi 0x7C1, 0x01");})
```

closed 2024-04-16 by Juan-Gg

jquevremont, 2024-06-10:
Have you looked at PMA? This is design-time configuration.

Note that the WT cache is due to evolve to HPDCache.

Juan-Gg, 2024-06-19:
> Have you looked at PMA? This is design-time configuration.
> 
> Note that the WT cache is due to evolve to HPDCache.

Thanks. My accelerator can read and write arbitrary regions of memory, so setting a region as non-cacheable would not cover the general case.