openhwfoundation/cva6 #2048: [BUG] Hardcoded 300s Timeout Issue during VCS Compile/RTL Simulation in cva6.py

xiaoweish, 2024-04-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'd like to bring to your attention an issue encountered while executing cva6.py. During script execution, a timeout error arises, as evidenced by the excerpt from the terminal:

```
cva6.py:137   INFO     Target: cv32a60x
cva6.py:138   INFO     ISA rv32imac_zba_zbb_z ...
cva6.py:604   INFO     [vcs-testharness] Running ...
lib.py:101   DEBUG    make vcs-testharness ...
lib.py:129   ERROR    Timeout[300s]: make vcs-testharness ...

```
Upon reviewing the log, it's evident that the timeout (set to 300s) occurs during the vcs compile for the vcs-testharness ISS configuration. Further investigation revealed that the timeout value is hardcoded within the run_c function in cva6.py(line 607).

Delving deeper, I found that the --gen_timeout option in cva6.py is utilized for the riscv-dv gen, while the --iss_timeout option is employed for the ISS (spike/whisper/ovpsim/...), as inferred from the riscv-dv/run.py file. However, it appears that cva6.py lacks a specific timeout setting for RTL, let's say --rtl_timeout.

I propose integrating such an option to enhance the functionality of cva6.py, providing more flexibility in handling various timeout scenarios. This approach would eliminate the necessity for hardcoded timeout settings (line 469/547/607) and empower users with greater control over script execution.

For reference, please see the figure at https://ibex-core.readthedocs.io/en/latest/_images/dv-flow.png, which illustrates the necessity for three timeout settings:

1. Timeout setting for make gen.
2. Timeout setting for make iss_sim.
3. Timeout setting for make compile,rtl_sim.

Thank you for considering this enhancement.

xiaoweish, 2024-04-17:
One more thing on cva6.py, 
Based on the logfile observation where cva6.py executes vcs-testharness before spike, I propose a rearrangement. Running spike first might be advantageous, as it's considerably faster. This adjustment could potentially expedite error detection, especially if ISG (RISCV-DV) generates any error codes. By prioritizing spike, we may catch these errors earlier in the testing process.

JeanRochCoulon, 2024-05-03:
Hello  @xiaoweish   Feel free to submit PRs. Of course as you speak about two different issues, better is to submit two different PRs. The second issue seems less relevant to me.

xiaoweish, 2024-05-13:
Hi @JeanRochCoulon , as to the second one, please check this PR: https://github.com/openhwgroup/cva6/pull/2104

xiaoweish, 2024-05-15:
Closed as fixed in above 2 PRs.

closed 2024-05-15 by xiaoweish