openhwfoundation/cva6 #2056: [BUG] External supervisor mode interrupt

AyoubJalali, 2024-04-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
This a CV32A65X RTL bug, as the core configuration does not support S-mode, so should not support S-mode interrupt, but the CV32A65X has an input irq_i[0] for Supervisor external interrupt, when irq_i[0] is assert the MIP.SEIP is asserted, that the normal behavior if we support the S-mode, but the CV32A65X does not.

Gchauvon, 2025-01-15:
I can't reproduce this issue. Asserting irq_i[0] only raise MIP.MEIP and MIP.SEIP stay at 0.
The RTL seems fine since PR #2017 which was merged few weeks before your issue.

Can we close this issue ?

AyoubJalali, 2025-01-15:
You should seert irq_i[1] so that you can see that MIP.SEIP is asserted.

PR #2716 (2025-01-21): csr_regfile: SEIP is read only 0 (fix #2056)
Fix #2056
commit 98604b59 (2025-01-21): csr_regfile: SEIP is read only 0 (fix #2056) (#2716)

Fix #2056

---------
closed 2025-01-21 by JeanRochCoulon