openhwfoundation/cva6 #2057: [BUG] The result of fsqrt.d is incorrect.

youzi27, 2024-04-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In some rounding mode, precision errors occur when calculating `fsqrt.d.`  And the results differ from those of Spike.
Testing has shown that RDN, RUP, RMM and DYN all experience rounding errors.

**How to reproduce and the logs are as follows:**
Example: 
Initialize `fa7 `with `0x402e000000000000` and execute `fsqrt.d ft10, fa7`.
Under `RDN, DYN` Mode:

Spike results are: `ft10 = 0x400efbdeb14f4eda`
CVA6 results are: `ft10 = 0x400efbdeb14f4ed9`

Under `RUP , RMM` Mode:
Spike results are: `ft10 = 0x400efbdeb14f4ed9`
CVA6 results are: `ft10 = 0x400efbdeb14f4eda`

youzi27, 2024-04-20:
The vulnerability of the rounding mode mentioned above also occurs in the `fdiv` instruction.

**Example:**
When executing the instruction `fdiv.d fs9, fs10, fa4` with the following register values:
```
fs10 = 0x41ddc16575c00000
fa4  = 0x41e0000c09000000
```
The results differ between CVA6 and Spike simulator as follows:
Round Toward Zero (**RTZ**) Mode:
CVA6 Result: `0x3fedc14f1407f47f`
Spike Result: `0x3fedc14f1407f47e` (correct value as verified)
Round Up (**RUP**), Round to Nearest, Tie to Max Magnitude (**RMM**), and Dynamic (**DYN**) Rounding Modes:
CVA6 Result: `0x3fedc14f1407f47e`
Spike Result: `0x3fedc14f1407f47f`

JeanRochCoulon, 2024-04-22:
This Github issue can also be reported on cvfpu side, thanks

youzi27, 2024-06-22:
Hi @Nolan-Ashford .

Thank you for your attention to this issue. 

However, in your response and the links you provided, I **did not** see any content related to the `DYN` and `RMM` rounding modes. Please carefully read the specific rounding modes mentioned in the issue I submitted.

Therefore, this is different from the previous issue and is a new bug.

Thank you for the discussion, but I hope you will conduct a more thorough review.