openhwfoundation/cva6 #2060: [BUG] Incorrect Result from 'fdiv' Operation Leading to Negative Infinity Instead of Smallest Negative Number

youzi27, 2024-04-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When performing floating-point division using the `fdiv.d` instruction in CVA6, under specific conditions where the expected result is the smallest representable negative double-precision floating-point number (`0xffefffffffffffff`), CVA6 incorrectly returns negative infinity (`0xfff0000000000000`). 

**Steps to Reproduce:**
1. Load `fa7` with `0xc024000000000000`.
2. Load `fs3` with `0x00000000000002cc`.
3. Execute the instruction: `fdiv.d ft6, fa7, fs3`.
4. Observe the resulting value in `ft6`.

**The log is as follows::**
```
core   0: 0x00000000800052b2 (0x233987d3) fsgnj.d fa5, fs3, fs3
1 0x00000000800052b2 (0x233987d3) f15 0x00000000000002cc
core   0: 0x00000000800052b6 (0x23188853) fsgnj.d fa6, fa7, fa7
1 0x00000000800052b6 (0x23188853) f16 0xc024000000000000
core   0: 0x00000000800052ba (0x1b38f353) fdiv.d  ft6, fa7, fs3
1 0x00000000800052ba (0x1b38f353) f 6 0xfff0000000000000
```

JeanRochCoulon, 2024-04-22:
This Github issue can also be reported on cvfpu side, thanks

youzi27, 2024-06-22:
Hi, @Nolan-Ashford .

Thank you for your attention to this issue. It's hard to say because it indeed corresponds to a special case. Moreover, if you are conducting verification, I suggest you try operations outside the rounding modes submitted to issue https://github.com/pulp-platform/fpu_div_sqrt_mvp/issues/14 (at least it is not completely consistent with the issue you submitted, so it is not considered a duplicate bug).  I hope you will conduct a careful and rigorous review when determining duplicates.

Additionally, at this point, I do not deny whether their root causes are the same (caused by the rounding mode). The reason I submitted this issue is that it corresponds to a special edge case, and in this case, I did not consider the impact of the rounding mode. I will review this issue again. 

Thank you.