openhwfoundation/cva6 #2065: [AXI dv Plan]: Minor feedback

ludovicpion, 2024-04-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,
As agreed, we (CEA) hae reviewed the AXI dv_plan. Only minor points to highlight.

Item: 013
•	XLEN is not mentioned in the specification. Do we have to understand the XLEN= AXI DATA WIDTH ?

Item: 014
In the dvPlan: ARSIZE can not be equal to 3 if ARID = 1 
but in the spec: if(RV32) ARSIZE != 3 && ARLEN = 0 && ARID = 1. 
=> So the ARLEN = 0 condition is missing in the dvplan

In the dvPlan: ARSIZE can not be equal to 0, 1 or 2 if ARLOCK = 1 
but in the spec if(RVA) AxLOCK = 1 => AxSIZE > 1. 
=> Should be 
ARSIZE can not be equal to 0 or 1 if ARLOCK = 1 (already present in the Item: 015)
ARSIZE is equal to 2 if ARLOCK = 1

Regards,
Ludovic

JeanRochCoulon, 2024-04-22:
Thank you @ludovicpion for this feedback

JeanRochCoulon, 2024-05-21:
@AEzzejjari   Do you confirm the feedbacks are inserted in the doc ?    If ok, the issue can be closed.

AEzzejjari, 2024-05-22:
Yes I do

closed 2024-05-22 by JeanRochCoulon