openhwfoundation/cva6 #2071: Purpose of AXI_BUS, axi_master_connect dependency with  axi_switch_vif in uvm_testbench

dvusingh, 2024-04-23:
Hi, 

I observed that in the **cva6_tb_wrapper.sv file**, 
1. we are connecting axi_interface signal based on axi_switch_vif as done in below assign statement
****assign axi_ariane_resp.aw_ready = (axi_switch_vif.active) ? axi_slave.aw_ready : cva6_axi_bus.aw_ready;**

Now again we are doing the same assignment, as **axi_switch_vif.active =1**
assign axi_slave.aw_ready  = (axi_switch_vif.active) ? axi_slave.aw_ready : cva6_axi_bus.aw_ready;**

we have AXI_BUS instance and axi_master_connect instance

 AXI_BUS.Master master
);
     assign master.aw_id         = dis_mem? '0 : axi_req_i.aw.id;
    assign master.aw_addr       = dis_mem? '0 : axi_req_i.aw.addr;

Here, master.aw_addr=0.

2. later we are connecting the axi2mem with sram for instruction and data storage

What is the purpose of connecting AXI_BUS and axi_master_connect with cva6 using axi_switch_vif?

I am not able to understand this can anyone explain.

Thanks

AEzzejjari, 2024-04-26:
Hello,

In our testbench, we have two slaves: the SRAM with AXI2MEM adapter connected to cva6_axi_bus, and the AXI slave agent connected to axi_slave. During simulation, we utilize only one slave.

The axi_switch_vif is created to toggle between the two slaves:

 - When the agent is in passive mode, we connect the SRAM to cva6, meaning cva6 receives input from cva6_axi_bus. In this scenario, the agent solely monitors all signals (thus, "assign axi_slave.aw_ready = (axi_switch_vif.active) ? axi_slave.aw_ready : cva6_axi_bus.aw_ready;").
 
 - When the agent is in active mode, it is responsible for setting the inputs of cva6 (hence, to switch between the two interfaces **assign axi_ariane_resp.aw_ready = (axi_switch_vif.active) ? axi_slave.aw_ready : cva6_axi_bus.aw_ready;).

dvusingh, 2024-05-03:
Hi, 

I ran few testcases, but I found that the axi_slave is always in active mode.
Also, I found 1 more issue that relates: Passive mode of AXI agents does not work anymore #1946

Any clarification on this.

Thanks

AEzzejjari, 2024-05-06:
Hi,

Like you said, there is a bug in the passive agent. Once it's fixed, you can use the agent in passive mode.

dvusingh, 2024-05-27:
Hi,

Any updates on the passive mode agent's issue.

Thanks

dvusingh, 2024-06-05:
Hi, 

I want to run my simulation with passive agents with SRAM support.
I want to collaborate to resolve the bug in passive agent in uvm-testbench. Any inputs how I should start to check the issue.

Thanks

AEzzejjari, 2024-06-05:
Hi,

The problem is fixed now. You can try using the passive mode
Thank you

dvusingh, 2024-06-06:
Hi @AEzzejjari ,

Thanks for the update.

So, I have to change the --axi_active=no and the list of files from this link https://github.com/openhwgroup/core-v-verif/commit/c01247b17eac00217784ef65f00ecbc87fe4ddbe.

Any other modifications required?

Thanks

dvusingh, 2024-06-07:
Hi @AEzzejjari ,

I took all the changes from the Connect the new AXI agent with CVA6 #2182 commit. Also added the --axi_active="no" in benchmark.sh file
When I am running the benchmark.sh (median test), then getting the below message and the then timeout occurs.

**median.cv32a6_imac_sv32.log.iss** : report
[TRACER] Output filename is: trace_hart_0.log
UVM_INFO @ 105.500 ns : uvmt_cva6_firmware_test.sv(159) uvm_test_top [TEST] Started RUN
UVM_INFO @ 783.500 ns : uvma_axi_covg.sv(265) uvm_test_top.env.axi_agent.axi_covg [uvma_axi_covg_c] READ ADDRESS ITEM DETECTED
UVM_INFO @ 783.500 ns : uvme_axi_covg.sv(279) uvm_test_top.env.cov_model.axi_covg [uvme_axi_covg_c] READ ADDRESS ITEM DETECTED
UVM_INFO @ 783.500 ns : uvme_axi_ext_covg.sv(272) uvm_test_top.env.cov_model.axi_ext_covg [uvme_axi_ext_covg_c] READ REQ ITEM DETECTED
UVM_INFO @ 789.500 ns : uvme_axi_covg.sv(285) uvm_test_top.env.cov_model.axi_covg [uvme_axi_covg_c] READ DATA ITEM DETECTED

Any suggestions on this issue?

Thanks

closed 2024-07-14 by github-actions[bot]