openhwfoundation/cva6 #2076: AXI UVM issue when integrating HPDCache

JeanRochCoulon, 2024-04-25:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When switching WT cache to HPD cache, 4 AXI assertions stop the simulations:
[image]

When commenting the 4 assertions, the hello word test stops on timeout after 300s. When switching the AXI from active to passive, the test passes.

This failure is using vcs-uvm target.
The Thales developing branch is hpdcache0

cfuguet, 2024-04-29:
Hello,

Thank you for the report.

Can you please send the command line to rerun your test ?

When I debugged one of the ADD assembly tests, I saw that the problem came effectively for a write, but what I saw, is that the HPDcache sent both the address (AW) and the data (W) but it never received a response for it (B). So, in my case it looks more like a problem from the test environment.

AEzzejjari, 2024-04-29:
When I reviewed the log file, I didn't notice that the agent didn't sent response, but when you mentioned it, I observed the waveform and I realized that when the address is accepted by the agent after the data, the agent does not send a response. Therefore, this is a bug in the agent.

cfuguet, 2024-04-29:
Ok, yes, that was what I observed too. Let me know if I can help.

cfuguet, 2024-05-02:
Hi @AEzzejjari,

Do you have any news on this issue ? :)

Thanks and regards

JeanRochCoulon, 2024-05-02:
I think @AEzzejjari is oOo Today.
Alae has fixed the AXI agent.
I am (trying) to bump core-v-verif to CVA6... Not sure to succeed soon, because two PRs are in collision.

cfuguet, 2024-05-02:
Ok great ! No problem, I'm glad that you have found some issues.

JeanRochCoulon, 2024-05-18:
fixed by #2125

closed 2024-05-18 by JeanRochCoulon