openhwfoundation/cva6 #2090: [BUG] <smoke-tests.sh fails to build Spike due to missing makefile>

ibooga, 2024-05-01:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Upon reaching step 6 of the quick setup instructions after entering:
export DV_SIMULATORS=veri-testharness,spike
bash verif/regress/smoke-tests.sh

I am told that Spike did not install due to missing makefiles.

Attache you will find the output of the failed installation.
[smoke-test_failure_spike.txt](https://github.com/openhwgroup/cva6/files/15172672/smoke-test_failure_spike.txt)

ibooga, 2024-05-01:
Spike failing due to missing file
[image]

ibooga, 2024-05-01:
It seems like within
g++ -MMD -MP  -DPREFIX=\"/home/cai/cache_project/cva6/tools/spike\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -DSPIKE_HASH_VERSION=0xdeb75451 -g -O2 -DPREFIX=\"/home/cai/cache_project/cva6/tools/spike\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -DSPIKE_HASH_VERSION=0xdeb75451 -std=c++17 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm -I/include -I//include -I/include -I/share/verilator/include/vltstd/ -I/usr/include   -c ../fesvr/fesvr_dpi.cc

That

-I/share/verilator/include/vltstd/

Is attempting to direct the compiler to where svdpi.h lives but is doing so unsuccessfully.

ibooga, 2024-05-01:
Started from scratch and Spike seems to build properly.

closed 2024-05-01 by ibooga

cllll402, 2024-05-17:
Can I get in touch with you more?  VX：aaaslla