openhwfoundation/cva6 #2091: Error in gcc-toolchain-builder build

cllll402, 2024-05-01:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I encountered a difficulty when building the gcc compiler toolchain, and I don't know how to solve it, can you help me?

I pre-installed the installation package required by the project, set the RISCV path, and obtained the source code of the tool chain.
[image]
[image]
[image]

But when I used the following command, I was told that the command format was wrong, so I used  sh build-toolchain.sh  gcc-13.1.0-baremetal $INSTALL_DIR command to build with the following error message:
[image]

g++ -std=gnu++11 -c   -O2     -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../../../src/gcc/gcc -I../../../src/gcc/gcc/build -I../../../src/gcc/gcc/../include  -I../../../src/gcc/gcc/../libcpp/include  \
	-o build/genrvv-type-indexer.o ../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc
../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc: In function ‘std::string same_ratio_eew_type(unsigned int, int, unsigned int, bool, bool)’:
../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:118:30: error: ‘log2’ is not a member of ‘std’
     elmul_log2 = lmul_log2 - std::log2 (sew / eew);
                              ^
../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:118:30: note: suggested alternative:
In file included from /usr/include/features.h:375:0,
                 from /usr/include/stdio.h:27,
                 from ../../../src/gcc/gcc/system.h:46,
                 from ../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:18:
/usr/include/bits/mathcalls.h:145:1: note:   ‘log2’
 __MATHCALL (log2,, (_Mdouble_ __x));
 ^
../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: ‘log2’ is not a member of ‘std’
     elmul_log2 = lmul_log2 + std::log2 (eew / sew);
                              ^
../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: note: suggested alternative:
In file included from /usr/include/features.h:375:0,
                 from /usr/include/stdio.h:27,
                 from ../../../src/gcc/gcc/system.h:46,
                 from ../../../src/gcc/gcc/config/riscv/genrvv-type-indexer.cc:18:
/usr/include/bits/mathcalls.h:145:1: note:   ‘log2’
 __MATHCALL (log2,, (_Mdouble_ __x));
 ^
make[2]: *** [build/genrvv-type-indexer.o] Error 1
make[2]: Leaving directory `/home/autumn/cva6/util/gcc-toolchain-builder/build/gcc/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/home/autumn/cva6/util/gcc-toolchain-builder/build/gcc'
make: *** [all] Error 2
*** Could not build GCC (even after removing target dirs), bailing out!

Environment: centos7

JeanRochCoulon, 2024-05-02:
Hello,
Reading your log, it seems the INSTALL_DIR is not setup. Before executing "sh build-toolchain.sh $INSTALL_DIR" can you "echo $INSTALL_DIR" ?

cllll402, 2024-05-02:
Thank you so much for answering my question.

[image]

I have set the paths to $RISCV and $INSTALL_DIR in my bashrc. Aside from this problem, are there any other factors that might cause a build to fail?

cllll402, 2024-05-09:
No error occurs after the INSTALL_DIR directory is specified as the cva6 peer directory and the centos gcc version 4.85 is upgraded to 9.3.1

closed 2024-05-12 by cllll402