openhwfoundation/cva6 #2096: Discrepancy between AxiIdWith and MemTIdWidth

JeanRochCoulon, 2024-05-03:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Two parameters exist to setup the ID, AxiIdWith and MemTIdWidth.
To me, they are equal. @cfuguet do you confirm ?

Currently IDwidth is set to 4. As AMO is not supported in CV32A65X, ID can be equal to 0 and 1, so IDwidth = 1. But if I set it to 1, hpdcache assertions are failed because of a parameter dependencies. Can ask you propose a fine-tuned parameter configuration ?

JeanRochCoulon, 2024-05-03:
@Gchauvon @ASintzoff @yanicasa

cfuguet, 2024-05-03:
Hello @JeanRochCoulon,

In a AXI-based System, yes, MemTidWidth shall be equal or less than AxiIdWidth.

The idea of the MemTidWidth was, if I remember well, to have a NoC agnostic parameter to define the width of the identifier used for transactions from the caches (instruction and data). For example, OpenPiton uses this parameter, and not the AxiIdWidth parameter, for obvious reasons.

Another thing is that the MemTidWidth is actually related to the capabilities of the caches. For example, the HPDcache is able to send up to WbufDirEntries outstanding write transactions, MshrWays*MshrSets read miss transactions, and one uncacheable write/read/AMO transaction. You need then at least MemTidWidth = max(log2(WbufDirEntries),log2(MshrWays*MshrSets)) + 1.

The AxiIdWidth is not related to the capabilities of the cache but of the entire system (or at least of a given interconnect in a given subsystem). You may need a wider AxiIdWidth than the MemTidWidth of the CVA6 if you have another initiator requiring more ID bits. That is why MemTidWidth must be less or equal to AxiIdWidth.

In your specific case, if you want MemIdWidth = 1, then you need to set WbufDirEntries=1, MshrWays=1 and MshrSets=1.

closed 2024-05-03 by JeanRochCoulon