openhwfoundation/cva6 #2099: How to Include apb_uart

dvusingh, 2024-05-03:
Hi,

We want to integrate the uart in cva6. We have integrated uart in cva6 the way it is shown in the ariane _peripherals.sv file. But we are not seeing any toggling on tx, rx pin of uart. Can we know how to include the apb_uart module in this ariane_tb.

****I found the dependency of VERILATOR variable to include the uart in ### corev_apu/tb/ariane_testharness.sv module**
line no 581

**`ifndef VERILATOR
    .InclUART     ( 1'b1                     ),
`else
    .InclUART     ( 1'b0                     ),
`endif**

Thanks

JeanRochCoulon, 2024-05-03:
To my mind, UART is avialable when targeting FPGA GenesysII board (to be confirmed).

dvusingh, 2024-06-05:
Hi,

Is there any testcase to verify the uart from vcs-testharness  or uvm simulator

Thanks

JeanRochCoulon, 2024-07-07:
No comment for a while. I close the issue, open another if needed

closed 2024-07-07 by JeanRochCoulon