openhwfoundation/cva6 #2106: There was a problem installing spike

cllll402, 2024-05-13:
The following error occurs when I run this step.
[image]
[image]
[image]
[image]

When I look up the tools path, I find that spike doesn't exist.
[image]

So I tried to run the install-spike.sh file separately, again encountering difficulties.
[image]

I checked the integrity of the file in the corresponding path.
[image]

What am I supposed to do now?

JeanRochCoulon, 2024-05-13:
Can you start from the first step, that is to say by cloning the cva6 repository?

cllll402, 2024-05-14:
Yes, I've tried that. I once again followed the instructions in step 1 to remove the project and pull the code repeatedly. I repeated the process, but there was still no spike in the /tools path.And just in case I missed something, I used cat cva6/.gitmodules to check each one.

cllll402, 2024-05-21:
Solve the problem after repeated pull item installation.

closed 2024-05-21 by cllll402

JeanRochCoulon, 2024-05-21:
Thanks for the feedback