openhwfoundation/cva6 #2112: AXI agent limitations

JeanRochCoulon, 2024-05-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

During the HPDCache, some limitations have detected in AXI agents, refer to #2097 :
- interleaving and out of order features are enabled/disabled by a single parameter which not relevant knowing some cache support only one of the two features. A second parameter is useful.
- interleaving and out of order features should be define as UV top parameter to be able to configure them in relation to the selected cache
- When integrating the HPDCache, some assert related to ID have been commented. This should be fixed.

@ASintzoff @AnouarZajni

cfuguet, 2024-05-16:
There other issues related to some illegal bins in coverpoints that are incompatible with the HPDcache. Hereafter you will find a patch commenting out these coverpoints. These bins are actually legal on AXI but they were tagged as illegal in your UVM environment because those ID values were not used by the other cache subsystems.

```diff
--- a/verif/env/uvme/cov/uvme_axi_covg.sv
+++ b/verif/env/uvme/cov/uvme_axi_covg.sv
@@ -72,11 +72,11 @@ covergroup cg_axi_b_channel(string name)
    option.per_instance = 1;
    option.name         = name;
 
-   bid:   coverpoint (item.b_id){
-      bins one   = {[1:3]};
-      illegal_bins ILLEGAL_BINS = {2};
-      ignore_bins  IGN_EXID = {3} iff(!RVA);
-   }
+   //bid:   coverpoint (item.b_id){
+   //   bins one   = {[1:3]};
+   //   illegal_bins ILLEGAL_BINS = {2};
+   //   ignore_bins  IGN_EXID = {3} iff(!RVA);
+   //}
    bresp: coverpoint (item.b_resp){
       bins zero  = {0};
       bins one   = {1};
@@ -84,7 +84,7 @@ covergroup cg_axi_b_channel(string name)
       bins three = {3};
    }   
 
-   b_axi_cross: cross bid, bresp;
+   //b_axi_cross: cross bid, bresp;
 endgroup : cg_axi_b_channel
 
 covergroup cg_axi_ar_channel(string name)
@@ -136,11 +136,11 @@ covergroup cg_axi_r_channel(string name)
    option.per_instance = 1;
    option.name         = name;
 
-   rid: coverpoint (item.r_data_trs[index].r_id) {
-      bins ID[] = {[0:3]};
-      illegal_bins ILLEGAL_BINS = {2};
-      ignore_bins  IGN_EXID = {3} iff(!RVA);
-   }
+   //rid: coverpoint (item.r_data_trs[index].r_id) {
+   //   bins ID[] = {[0:3]};
+   //   illegal_bins ILLEGAL_BINS = {2};
+   //   ignore_bins  IGN_EXID = {3} iff(!RVA);
+   //}
 
    rlast: coverpoint (item.r_data_trs[index].r_last);

@@ -151,9 +151,9 @@ covergroup cg_axi_r_channel(string name)                                                         
       bins three = {3};
    }

-   r_axi_cross: cross rid, rlast, rresp {
-   illegal_bins ILLEGAL_BINS = binsof(rid) intersect{3} && binsof(rlast) intersect{0};
-   }
+   //r_axi_cross: cross rid, rlast, rresp {
+   //illegal_bins ILLEGAL_BINS = binsof(rid) intersect{3} && binsof(rlast) intersect{0};
+   //}
 endgroup : cg_axi_r_channel

 /**

JeanRochCoulon, 2024-05-18:
fixed by #2125

closed 2024-05-18 by JeanRochCoulon