openhwfoundation/cva6 #2116: [BUG] Linux boot on Nexys Video CVA6 64-bit configuration

amiroshni, 2024-05-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Red Semi is trying to get Linux running on the Nexys Video build of the CVA soc, and this follows on the work from https://github.com/openhwgroup/cva6/pull/1925

Digilent page on the [Nexys Video board](https://digilent.com/reference/programmable-logic/nexys-video/start)
FPGA model: XC7A200T-1SBG484C

The current work done by @Saute0212 allows to boot into Linux, however the ethernet controller is disabled on the FPGA design.

As part of the exercise, my colleague and I have been documenting the steps required to get to running Linux on FPGA: https://github.com/amiroshni/documenting-cva6

*(Apologies to the CVA6 devs for adding clutter to the previously completed PR.)*

Copying my last comment on the progress we had in Red on getting the FPGA to boot Linux:

Thank you for the fast response @Saute0212 !

My colleague and I followed your suggestions (except for the 4GB microsd card, but we ordered some).

> @amiroshni Thanks for contacting me. I have confirmed that the version of bootrom is correct.
> 
> I have three questions.
> 
>     * Did you run `make fpga` after running `make clean`?
Yes, we did.

> 
>     * Is it possible to reduce the capacity of the microSD card? For example, 4GB.
Currently we only have the 16GB cards on hand, but ordered some 4GB ones for testing. Should arrive in 3-5 days.

> 
>     * Could you try to clone the repository I am forking and generate Bitstream?
>       CVA6 repository I forked : https://github.com/Saute0212/cva6
Yes. Changed my documentation to use your cva6 fork: https://github.com/amiroshni/documenting-cva6

My colleague @shriyasharma11 generated the new bitstream and created a serial log of the boot process here: https://gist.github.com/shriyasharma11/12136b710edaa7eac2e1269a903f8cef

Interesting problems seem to repeat throughout the kernel boot:
```
[   39.968943] mmc0: new SDHC card on SPI
[   40.105682] mmcblk0: mmc0:0000 SC16G 14.8 GiB 
Starting logging: [   40.380820]  mmcblk0: unable to read partition table
[   40.419800] mmc0: SPI card removed
[   41.148149] mmc0: host does not support reading read-only switch, assuming write-enable
[   41.170358] mmc0: new SDHC card on SPI
[   41.380054] mmcblk0: mmc0:0000 SC16G 14.8 GiB 
[   41.731772]  mmcblk0: unable to read partition table
[   41.835341] mmc0: SPI card removed
OK
[   42.674518] mmc0: error -110 whilst initialising SD card
```
(even though the micro sd card is not touched, kernel seems to think it's being removed).

We will also try on a different 16GB card (we have three of those), and once the 4GB arrives will test as well.

Edit: I also forgot to mention. After a while this error persists, and there is no further progress:

```
mmc0: error -110 whilst initialising SD card
```

Further CPU resets don't make a difference (no more response from the SD card). My colleague actually managed to get the kernel to boot like this yesterday, but we needed to re-generate the bitstream today get a good plaintext log of the kernel. I wonder if there's also an FPGA-related issue (maybe timing) which stops the SD card from being read?

_Originally posted by @amiroshni in https://github.com/openhwgroup/cva6/issues/1925#issuecomment-2112218421_

Since my last comment, we were able to get the bootrom to detect and copy the kernel from the SD card by unplugging and plugging in the micro sd card again (looks like the Nexys Video might have issue with our micro SD cards).
I will update once we test using 4GB micro SD cards.

Saute0212, 2024-05-16:
@amiroshni 
Thank you for contacting me.
Please let me know if you know the result of changing the capacity of the microSD card to 4GB.
Thank you.

amiroshni, 2024-05-20:
> Please let me know if you know the result of changing the capacity of the microSD card to 4GB. Thank you.

Thank you very much for your help @Saute0212 !

Turns out the 4GB card works!

My colleague generated this log, showing that we can login as root and access the standard utilities:
https://gist.github.com/shriyasharma11/8afee5a594f2f05512551a72740426cd

To anyone else following, with this guide, you can setup Nexys Video for yourself (just bear in mind support is experimental, and not as complete as the Genesys 2 FPGA board):
https://github.com/amiroshni/documenting-cva6

Now we need to think how to load programs without a network connection.
If we wanted to send a binary over, can it be done via the serial connection?

Saute0212, 2024-05-21:
@amiroshni 
I'm glad you were able to boot Linux with Nexys Video.

Also, thanks for your documentation on the Nexys Video setup.

I am also looking for a way to load the program without a network connection. 
It would be useful to be able to send binaries over a serial connection.

Thank you.