openhwfoundation/cva6 #2119: [BUG] <bitstream generated for cv32a6_ima_sv32_fpga does not boot Linux>

AngelaGonzalezMarino, 2024-05-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi @Saute0212 @JeanRochCoulon

I guess an issue is more appropriate for this discussion

After PR https://github.com/openhwgroup/cva6/pull/1925  the bitstream generated for cv32a6_ima_sv32_fpga does not boot Linux.
I tried on Genesys 2, but I think it will be the same for Nexys Video

The issue is that the makefile for bootrom generation now uses
-march=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei which assumes that a few extensions are always there

In the case of cv32a6_ima_sv32_fpga the C extension (and others) is not there, therefore it never boots

Given that extensions are optional, I think we should keep the simplest march, to have a common bootrom compilation and avoid this kind of problem. What do you think?

I have been able to successfully generate a bitstream and boot correctly on Genesys2 with the following configurations:

    cv32a6_ima_sv32_fpga using -march=rv32im_zicsr
    cv64a6_imafdch_sv39 using -march=rv64im_zicsr

Saute0212, 2024-05-16:
@AngelaGonzalezMarino 
Thank you for your suggestion.
I agree to keep the simplest march.
And I agree to change the ```Makefile``` in ```cva6/corev_apu/fpga/src/bootrom/``` as you suggested.
Thank you.

AngelaGonzalezMarino, 2024-05-16:
fixed in PR https://github.com/openhwgroup/cva6/pull/2121

closed 2024-05-16 by AngelaGonzalezMarino