openhwfoundation/cva6 #2122: ZEXT.H RV64 encoding causes an illegal instruction trap

LQUA, 2024-05-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

According to the specification, ZEXT.H instruction has a RV64 encoding definition : 

[image]

In the rtl (decoder.sv), ZEXT.H - RV64 encoding is not available : 
```
        // --------------------------
        // 32bit Reg-Reg Operations
        // --------------------------
        riscv::OpcodeOp32: begin
          instruction_o.fu = (instr.rtype.funct7 == 7'b000_0001) ? MULT : ALU;
          instruction_o.rs1[4:0] = instr.rtype.rs1;
          instruction_o.rs2[4:0] = instr.rtype.rs2;
          instruction_o.rd[4:0] = instr.rtype.rd;
          if (CVA6Cfg.IS_XLEN64) begin
            unique case ({
              instr.rtype.funct7, instr.rtype.funct3
            })
              {7'b000_0000, 3'b000} : instruction_o.op = ariane_pkg::ADDW;  // addw
              {7'b010_0000, 3'b000} : instruction_o.op = ariane_pkg::SUBW;  // subw
              {7'b000_0000, 3'b001} : instruction_o.op = ariane_pkg::SLLW;  // sllw
              {7'b000_0000, 3'b101} : instruction_o.op = ariane_pkg::SRLW;  // srlw
              {7'b010_0000, 3'b101} : instruction_o.op = ariane_pkg::SRAW;  // sraw
              // Multiplications
              {7'b000_0001, 3'b000} : instruction_o.op = ariane_pkg::MULW;
              {7'b000_0001, 3'b100} : instruction_o.op = ariane_pkg::DIVW;
              {7'b000_0001, 3'b101} : instruction_o.op = ariane_pkg::DIVUW;
              {7'b000_0001, 3'b110} : instruction_o.op = ariane_pkg::REMW;
              {7'b000_0001, 3'b111} : instruction_o.op = ariane_pkg::REMUW;
              default: illegal_instr_non_bm = 1'b1;
            endcase
            if (CVA6Cfg.RVB) begin
              unique case ({
                instr.rtype.funct7, instr.rtype.funct3
              })
                // Shift with Add (Unsigned Word)
                {7'b001_0000, 3'b010}: instruction_o.op = ariane_pkg::SH1ADDUW; // sh1add.uw
                {7'b001_0000, 3'b100}: instruction_o.op = ariane_pkg::SH2ADDUW; // sh2add.uw
                {7'b001_0000, 3'b110}: instruction_o.op = ariane_pkg::SH3ADDUW; // sh3add.uw
                // Unsigned word Op's
                {7'b000_0100, 3'b000}: instruction_o.op = ariane_pkg::ADDUW;    // add.uw
                // Bitwise Shifting
                {7'b011_0000, 3'b001}: instruction_o.op = ariane_pkg::ROLW;     // rolw
                {7'b011_0000, 3'b101}: instruction_o.op = ariane_pkg::RORW;     // rorw
                default: illegal_instr_bm = 1'b1;
              endcase
              illegal_instr = illegal_instr_non_bm & illegal_instr_bm;
            end else begin
              illegal_instr = illegal_instr_non_bm;
            end
          end else illegal_instr = 1'b1;
        end
```

I suggest adding the line for handling ZEXTH
```                // Bitwise Shifting
                {7'b011_0000, 3'b001}: instruction_o.op = ariane_pkg::ROLW;     // rolw
                {7'b011_0000, 3'b101}: instruction_o.op = ariane_pkg::RORW;     // rorw
                // Zero Extend Op
                {7'b000_0100, 3'b100} : instruction_o.op = ariane_pkg::ZEXTH;
                default: illegal_instr_bm = 1'b1;
              endcase
```

AyoubJalali, 2024-05-16:
I don't know which version of CVA6 your using by I have this in the decoder.sv in the last commit : https://github.com/openhwgroup/cva6/blob/9142fdd03a3d333235b7cb03c74f0d9cea177d25/core/decoder.sv#L857

LQUA, 2024-05-16:
Please forgive me, I'm new to this project (and GitHub).
I agree, the issue is fixed by [this commit](https://github.com/openhwgroup/cva6/commit/9bd56679928f68a4f7ea0b31486fb143cbb44694)

closed 2024-05-16 by LQUA