openhwfoundation/cva6 #2128: Error in `make fpga` of official release cva6-5.0.1

0ena, 2024-05-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I have downloaded the latest release of the CVA6 source code from here: https://github.com/openhwgroup/cva6/releases/tag/v5.0.1

and I am trying to do a test run by creating a bitstream, by running
`make fpga`

However the command fails, complaining about missing files 

```
Makefile:46: must set CVA6_REPO_DIR to point at the root of CVA6 sources -- doing it for you...
Makefile:143: XCELIUM_HOME not set which is necessary for compiling DPIs when using XCELIUM
Traceback (most recent call last):
  File "util/flist_flattener.py", line 75, in <module>
    parseFlist(args.inFlist, args.outFlist, args.print_incdir, args.print_newline)
  File "util/flist_flattener.py", line 41, in parseFlist
    raise (RuntimeError(f'{includedFilename} not found'))
RuntimeError: /MY_PATH/cva6-5.0.1/core/cache_subsystem/hpdcache/rtl/hpdcache.Flist not found
make: *** No rule to make target /MY_PATH/cva6-5.0.1/corev_apu/register_interface/src/reg_intf.sv', needed by 'fpga'.  Stop.
```

Is this supposed to happen with the official release code?
How can I fix this?
I am using the CVA6 for a project I am building and I would like to tie my project to the latest official release of CVA6.

Thanks!

JeanRochCoulon, 2024-05-18:
This file belongs to a CVA6 submodule. Are you sure the submodule are updated ?   Try to execute "git submodule update" in you cva6 repository.

0ena, 2024-05-19:
Hi @JeanRochCoulon ,

thank you for your response.
I tried your suggestion and got the following error:

```
fatal: Not a git repository (or any parent up to mount point /MY_MOUNTPOINT/)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

I then executed `git init` and then `git submodule update`, which completed without an issue but I still get the same errors.
Looks like the submodules are not getting updated for some reason. Any ideas why?

Thank you for your support.

JeanRochCoulon, 2024-05-19:
The error tells you you are not in a git repository. the command "git submodule update" need to be executed in cva6 repository !

0ena, 2024-05-19:
Hi @JeanRochCoulon ,

as I said I downloaded the tarbal from: https://github.com/openhwgroup/cva6/releases/tag/v5.0.1
I then decompress the tarbal and `cd` inside the CVA6 folder.
Then I execute all the above mentioned commands.

Kind reagrds,
Nassos

JeanRochCoulon, 2024-05-19:
@ASintzoff   I fear that the submodule are not included in the release !?

ASintzoff, 2024-05-20:
The tarball file contains only files belonging to cva6 repository but not the git repository itself. Therefore no git commands will succeed as there is no git repository in the decompress tarball.

To be able to perfom `make fpga`, the only way to proceed is to clone the cva6 repository as described in the [README](https://github.com/openhwgroup/cva6/blob/master/README.md).

```
git clone https://github.com/openhwgroup/cva6.git
cd cva6
git submodule update --init --recursive
```

JeanRochCoulon, 2024-05-21:
This issue seems closed. Open another if needed.

closed 2024-05-21 by JeanRochCoulon