openhwfoundation/cva6 #2129: [BUG] Incorrectly set fflags: An underflow exception is triggered when the computed result is an exact subnormal number.

fly-1011, 2024-05-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When using the division instruction, CVA6 incorrectly triggers an `underflow` exception when the computed result is an `exact subnormal number` (e.g., `0x00000001`). This contradicts the IEEE 754 standard, which specifies that an exact subnormal result should not trigger the underflow exception flag.

Steps to Reproduce:

1. Initialize ft4 to `0x00000001` and ft6 to `0x3F800000` 
2. Execute the instruction: `fdiv.s ft1, ft4, ft6`.
3. Observe the value of fflags.

The log from CVA6 is as follows:
```
core   0: 0x000000008000201e (0x186270d3) fdiv.s  ft1, ft4, ft6
3 0x000000008000201e (0x186270d3) f 1 0xffffffff00000001
core   0: 0x0000000080002022 (0x001023f3) csrrs   t2, fflags, zero
3 0x0000000080002022 (0x001023f3) x 7 0x0000000000000002
```

The log from Spike is as follows:
```
core   0: 0x000000008000201e (0x186270d3) fdiv.s  ft1, ft4, ft6
core   0: 3 0x000000008000201e (0x186270d3) f1  0xffffffff00000001
core   0: 0x0000000080002022 (0x001023f3) csrrs   t2, fflags, zero
core   0: 3 0x0000000080002022 (0x001023f3) x7  0x0000000000000000
```
The computed result is `2^-149`, which can be exactly represented as a subnormal number. Therefore, in this case, the underflow exception should not be triggered.

I will also report this issue to CVFPU.

fly-1011, 2024-05-19:
> This issue is a duplicate of [openhwgroup/cvfpu#61](https://github.com/openhwgroup/cvfpu/issues/61)
> 
> Additionally, if you want to test CVFPU, I kindly suggest testing it independently from CVA6 because, based on my observation, they are not always synchronized.

Hi @Codemaker-1 ,
Thanks again for your comments and suggestions. It appears that you are not a contributor to cva6 and have not delved deeply into this issue. 

 The issue you mentioned at https://github.com/openhwgroup/cvfpu/issues/61 indicates that when executing the fmul, fmadd, fmsub, fnmadd, and fnmsub instructions, if the result is a subnormal number, the UF flag is set to 1. But the issue I raised was that when the div instruction was executed, the UF flag was incorrectly set when the result was an exact subnormal number. So this is a new issue.
 
 Additionally, the issue you mentioned was raised in 2022, and CVFPU has been updated in 2023. For details, see: https://github.com/openhwgroup/cva6/issues/1071#issuecomment-1510467510.
 
 So this is a new bug.

MikeOpenHWGroup, 2024-05-19:
Hi @fly-1011, thanks for creating this issue (and the associated [issue on cvfpu](https://github.com/openhwgroup/cvfpu/issues/121)).  As @Codemaker-1 says, the version of the CVA6 is not up-to-date with the latest version of cvfpu.  This is largely due to the fact that the team is focused on the [CV32A65X](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x_design/source/index.html) which will be the first formal release of a fully verified core based on the CVA6.  The CV32A65X does not support floating point.

Would you be willing to join the CVA6 project meeting to discuss this?  The next meeting is 2024-05-28 at 16h00 CEST.  If you can join us that would be great and I will work with the team to get this issue on the agenda of that meeting.

fly-1011, 2024-05-20:
> Hi @fly-1011, thanks for creating this issue (and the associated [issue on cvfpu](https://github.com/openhwgroup/cvfpu/issues/121)). As @Codemaker-1 says, the version of the CVA6 is not up-to-date [...]

Thank you very much for your reply，

Thank you very much for your meeting invitation. Unfortunately, due to personal reasons, I will not be able to attend the meeting. However, I would greatly appreciate it if this issue could be included in the agenda for discussion. Thank you once again for your understanding and support!