openhwfoundation/cva6 #2136: [BUG] Cadence Xcelium dependent checksum verilog_package doesn't match

yunuseryilmaz18, 2024-05-21:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, I'm trying to run ariane_tb.sv on Cadence Xcelium 22.03 from `xrun_all` Makefile target. I've modified `ariane_pkg` variable for missing packages and compiled cva6 successfully. However, I can't run ariane as the top module, Xcelium gives this error:

```
xmelab: *E,DLCSMD: Dependent checksum verilog_package worklib.axi_pkg:sv (VST) doesn't match with the checksum that's in the header of: module worklib.ariane_testharness:sv (VST).
    ) dut (
        |
xmelab: *E,CUVMUR (../corev_apu/tb/ariane_tb.sv,77|8): instance 'ariane_tb.dut' of design unit 'ariane_testharness' is unresolved in 'worklib.ariane_tb:sv'.
```

I've used `axi_pkg.sv` package in the `vendor/pulp-platform/axi/src/axi_pkg.sv`. Did anyone encounter this error? Can you help me?

JeanRochCoulon, 2024-05-21:
@CoralieAllioux

CoralieAllioux, 2024-06-14:
hi @yunuseryilmaz18, sorry for the late answer.
We're working on the support of Cadence Xcelium simulator. Those days, some PR are ongoing to do so. This PR when merged could resolved your problem: https://github.com/openhwgroup/cva6/pull/2223

yunuseryilmaz18, 2024-06-28:
@CoralieAllioux I'm sorry too, we were on a national holiday and couldn't look into it. I've pulled from the repo today and run with Makefile again. The error I got:

```bash
file: cva6/core/cache_subsystem/cva6_hpdcache_wrapper.sv
    parameter hpdcache_pkg::hpdcache_cfg_t HPDcacheCfg,
                                                      |
xmvlog: *E,SVVMAP (/home/yunus.eryilmaz/gitlab_repo/cva6/core/cache_subsystem/cva6_hpdcache_wrapper.sv,20|54): Omitting the assignment to constant_param_expression in a param_assignment in a parameter_port_list currently is not supported.
```

Thank you for your answer.

yunuseryilmaz18, 2024-07-19:
Hi @CoralieAllioux,
I've added these variables to my environment:
```
export CADENCE_ENABLE_AVSREQ_44905_PHASE_1=1
export CADENCE_ENABLE_AVSREQ_63188_PHASE_1=1
```
We use Xcelium 23.09.003, I'm guessing you're using an older version of Xcelium and this results in different outcomes. Now I have this error:
```
`USER_MEM(load_address) = mem_row;
                                                   |
xmelab: *E,CUVNAA (../corev_apu/tb/ariane_tb.sv,158|51): An index has been applied to an inappropriate type.
```

Thank you for your answers.

CoralieAllioux, 2024-07-25:
Hi @yunuseryilmaz18,

We use on our side Xcelium 23.09.005. The PR https://github.com/openhwgroup/cva6/pull/2389 which is now merged should resolve your issue.

JeanRochCoulon, 2024-07-25:
Nice @CoralieAllioux !
@yunuseryilmaz18 can you close the issue if working on your side?