openhwfoundation/cva6 #2172: [BUG] Spike unknown version issue

shreyas-kalikar, 2024-05-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have been trying to set everything up by following the instructions on the repo, I want to be able to run Hello World on the CVA6. However, when I get to the following step:

`python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"`

I get the following error: 

`Thu, 30 May 2024 17:10:40 INFO     GCC Version: 13.1.0 `

`Thu, 30 May 2024 17:10:40 ERROR    You are currently using version - unknown - of Spike, should be: 1.1.1-dev e823d83. Please install or reinstall it with the installation script.`

I have tried looking this error up, did not come across anything familiar. I am unsure whether it is even able to access spike (which is why it says "version - unknown-", or if I do have spike, but just the wrong version, and the version check for some reason isn't able to tell me what my current version of Spike is.

For background, I have set SPIKE_PATH as: `../cva6/tools/spike/bin/spike`

Please advise me on how I can get around this, thank you!

JeanRochCoulon, 2024-05-31:
Please rebase your cva6 env, recommendations have been added to help cleaning the env before executing cva6.py

sahilswaroop99, 2024-06-03:
I have tried the troubleshooting of the issues of spike version 
there are couple issues persisting
Is there any specific environmental cleanups? recommendations you mention?

JeanRochCoulon, 2024-06-03:
If you use the latest version of CVA6, with the error Spike message your will have the recommendations to solve the env, that is to say to remove two build directories.

shreyas-kalikar, 2024-06-04:
Thank you for all the input @JeanRochCoulon! I hadn't gotten the build cleanup log prompt even on the newest version of the code, but I think that was because there was something wrong with the Spike installation directory for some reason (had to move it into the tools folder again) and it just said Spike version -{ unknown}. However, fixing the directory is now giving displaying the Spike version but giving me a new error, currently looking into it. I will raise a separate ticket for it. I appreciate it a lot, thank you!

closed 2024-06-04 by shreyas-kalikar