openhwfoundation/cva6 #2188: [BUG] Issue with Running Hello World C Program in CVA6

shreyas-kalikar, 2024-06-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I am encountering an issue when attempting to get hello world running in CVA6. The simulation fails with an error, and I am unable to proceed. Below are the details of the error message:

`Mon, 03 Jun 2024 16:30:53 INFO     GCC Version: 13.1.0`
`Mon, 03 Jun 2024 16:30:53 INFO     Spike Version: 1.1.1-dev b92d30f`
`Mon, 03 Jun 2024 16:30:53 INFO     Verilator Version: Verilator 5.008 2023-03-04 rev v5.008`
`Mon, 03 Jun 2024 16:30:53 INFO     Creating output directory: /home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03`

`Mon, 03 Jun 2024 16:30:53 INFO     Iteration number: 1`
`Mon, 03 Jun 2024 16:30:53 INFO     Compiling c test: ../tests/custom/hello_world/hello_world.c`
`Mon, 03 Jun 2024 16:30:53 INFO     Converting to /home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/directed_c_tests/hello_world.bin`
`Mon, 03 Jun 2024 16:30:53 INFO     Processing ISS setup file: cva6.yaml`
`Mon, 03 Jun 2024 16:30:53 INFO     Found matching ISS: veri-testharness`
`Mon, 03 Jun 2024 16:30:53 INFO     Target: cv32a60x`
`Mon, 03 Jun 2024 16:30:53 INFO     ISA rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei`
`Mon, 03 Jun 2024 16:30:53 INFO     [veri-testharness] Running ISS simulation: make veri-testharness target=cv32a60x variant=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/directed_c_tests/hello_world.o path_var=/home/shreyask/arails/cva6_shreyas/cva6/ tool_path=/home/shreyask/arails/cva6_shreyas/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/veri-testharness_sim/hello_world.cv32a60x.log &> /home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/veri-testharness_sim/hello_world.cv32a60x.log.iss`
`Mon, 03 Jun 2024 16:30:59 INFO`     
`Mon, 03 Jun 2024 16:30:59 ERROR    ERROR return code: True/2, cmd:make veri-testharness target=cv32a60x variant=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/directed_c_tests/hello_world.o path_var=/home/shreyask/arails/cva6_shreyas/cva6/ tool_path=/home/shreyask/arails/cva6_shreyas/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/veri-testharness_sim/hello_world.cv32a60x.log &> /home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/veri-testharness_sim/hello_world.cv32a60x.log.iss`

This is when I execute the following command:

`python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"`

I am using only veri-testharness in `$DV-SIMULATORS`.
I have reviewed a similar issue on the repository (Issue#1631), which suggested reinstalling the toolchain according to the repository's instructions. Unfortunately, this did not resolve the problem. Any guidance on how to resolve this issue would be greatly appreciated. If further information or logs are needed, please let me know!

JeanRochCoulon, 2024-06-04:
The error is in /home/shreyask/arails/cva6_shreyas/cva6/verif/sim/out_2024-06-03/veri-testharness_sim/hello_world.cv32a60x.log.iss, to be analyzed.

shreyas-kalikar, 2024-06-05:
Hi @JeanRochCoulon, sorry for the delay, I am still working on it and report my findings as soon as possible. I want to provide some solid findings so I will collect all the information and put them up. Thank you!

shreyas-kalikar, 2024-06-27:
Closing this issue, my sourced paths were jumbled. Have raised new issue regarding new error, thank you!

closed 2024-06-27 by shreyas-kalikar