openhwfoundation/cva6 #2192: Read-only 0 CSRs

JeanRochCoulon, 2024-06-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

MHPMEVENT wrong legal value:
[image]

MHPCOUNTERS/H wrong legal value:
[image]

MCYCLE/H, MINSTRET/H are RW registers and not WARL
[image]

MTVEC is RW CSR
[image]

The MCAUSE legal_values are wrong, to be discussed with  @ASintzoff
[image]

The MTVAL CSR is read only 0, so ROCST
[image]

zchamski, 2024-06-10:
In #2206 the `riscv-config` spec was adjusted so that after applying the doc generation script:
- `MTVAL` is RO_CONSTANT zero.
- `MHPMEVENT*`, `MHPCOUNTER*` and MCOUNTINHIBIT` are RO_CONSTANT zero.

For the legal value range of `MCYCLE`/`MCYCLEH` we need to adjust the presentation than the spec.

`Handling of bitmasked fields like `MTVEC` (in this case: bit zero always ignored on write, always cleared on read), the rendering needs additional information that is currently dropped when parsing input coming from `riscv-config`.

JeanRochCoulon, 2024-06-11:
#2231 fixed MTVAL RTL issue

JeanRochCoulon, 2024-06-14:
These above modifications have been added to the CSR specification. The PR can be closed.

closed 2024-06-14 by JeanRochCoulon