openhwfoundation/cva6 #2198: Bug : Spike raise an exception on reading some legal CSRs

AyoubJalali, 2024-06-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, Spike raise exception on reading from hpmcounter3-31h , these CSRs are implemented in CV32A65X and it's RO-ZERO CSRs.

[image]

MarioOpenHWGroup, 2024-06-05:
The `zihpm` is not enabled as far as I know, we should enable it in spike if we want it

JeanRochCoulon, 2024-06-05:
The Zicnt and Zihpm are not supported by 65x, but the CSRs referenced by @AyoubJalali are  MHPMEVENT and MPPMCOUNTERS/H at 0xC** addresses (there are not part of the previous listed extensions).

[image]

zchamski, 2024-06-05:
As far as I can tell from reading the RTL, the `Zihpm` extension ***is*** implemented: I see both ***M***HPMCOUNTER*n* CSRs (range 0xB03..0xB9F) and those without the leading ***M*** (range 0xC03..0xC9F), cf. https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv#L669 and ff.

The privilege check on the 0xC.. CSRs says 'no violation' if the privilege mode is M (https://github.com/openhwgroup/cva6/blob/master/core/csr_regfile.sv#L2125).

JeanRochCoulon, 2024-06-05:
Zihpm does not correpond to 0xB03..0xB9F  and range 0xC03..0xC9F !!   But to 0xC.. addresses. 
Please @zchamski confirm it.

AyoubJalali, 2024-06-05:
> The Zicnt and Zihpm are not supported by 65x, but the CSRs referenced by @AyoubJalali are MHPMEVENT and MPPMCOUNTERS/H at 0xC** addresses (there are not part of the previous listed extensions). [...]

Just to confirm spike support MHPMEVENT and MPPMCOUNTERS/H, i'm issue talk about hpmcounter[n]3-31h form 0xC03 to 0xC1F and from 0xC83 to 0xC9F

JeanRochCoulon, 2024-06-05:
Spike seems aligned with 65x specification, but RTL is not aligned. RTL implements Zihpm, but i should not. RTL need to be fixed.

AyoubJalali, 2024-06-05:
so this is an RTL issue @JeanRochCoulon you can confirm this to create an issue in the CVA6 and close this

JeanRochCoulon, 2024-06-05:
This gihub issue is superseeded by #2201

closed 2024-06-05 by JeanRochCoulon