openhwfoundation/cva6 #2199: Some confusion about using Verdi to view.vcd files

cllll402, 2024-06-05:
Following the instructions, I have successfully generated the.vcd waveform file for the script smoke-tests.sh.

But when I opened the waveform file and wanted to analyze it in modules, I was confused. I don't know where to start, and there don't seem to be any guidelines in the documentation that mention this part.

Here are some of the attempts I made and the difficulties I encountered. I would appreciate it if anyone could give me some guidance.

First, I convert the.vcd file to the.fsdb file.
[image]

Second, I reconfigured the Verdi command parameters and set the environment variables needed in the file Flist.cva6.
[image]
[image]
[image]

Finally, I use Verdi to open the file in the.vcd build directory.
[image]

The error appears as follows:
[image]
[image]
[image]
Is there anything else I didn't set up properly?

JeanRochCoulon, 2024-06-05:
To use VERDI,     "export VERDI=1" before running the simulation.
Please close the issue if solved.

cllll402, 2024-06-05:
Thank you for your answer. I enabled "export VERDI=1", but the problem still persists.

JeanRochCoulon, 2024-06-05:
You should open verdi by executing cva6.py, is it the case on your side ?

JeanRochCoulon, 2024-06-10:
No updates... I close the github issue, re-open it if needed.

closed 2024-06-10 by JeanRochCoulon