openhwfoundation/cva6 #2200: Failed to run any other C programs other than hell_world.c

Tanishqgithub, 2024-06-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I am constantly trying to run many other C programs other than hello_world.c program specified in the repository(which is program to print hello world! ) and i can see that only the C programs which uses <stdio.h> only gets successfully executed. 
like this progam : binary_sort.c  
[binary_sort.txt](https://github.com/user-attachments/files/15587481/binary_sort.txt)

And then i tried with other programs which uses other libraries such as <string.h> ; <stdbool.h> ; <stdlib.h> like these programs : reverse_string.c  
[image]
and some more examples which contain malloc() {like Huffman Coding examples which is there in  
[huffman_coding.txt](https://github.com/user-attachments/files/15587722/huffman_coding.txt)
}

The above programs are not successfully running and produce this log file : 
[hello_world.cv32a60x.log.iss_for_issue.txt](https://github.com/user-attachments/files/15587546/hello_world.cv32a60x.log.iss_for_issue.txt)
and even it is failing the smoke test (not the other benchmarks but the test where hello_world.c is tested)

I don't know if i am following the correct procedure to execute these programs or is there any issue regarding this. Please help me out.
This is what i did to run any other C program : 
1. change the contents of hello_world.c to whichever C program i need to run 
2. run the smoke test 
3. then use the classical command to run hello_world.c in processor 
i,e 
[image]
Please clarify me if i am correct regarding the steps followed to run C programs or is there any additional steps that has to included to achieve successful execution 

And please can u give an example command to run the Assembly level Programs in this processor

JeanRochCoulon, 2024-06-05:
Perhaps analyzing coremark or dhrystone benchmarks which are included in the CI canhelp you. They are in C and dayly run on CVA6.

Tanishqgithub, 2024-06-06:
As specified by you i tried to run the dhrystone benchmarks and got this log file : 
[image]
and the corresponding log file : 
[dhrystone_main.hwconfig.log.iss.txt](https://github.com/user-attachments/files/15601721/dhrystone_main.hwconfig.log.iss.txt)
i tried to analyze the trace file but couldn't get any conclusions from it .

i even tried to run all the benchmarks using the file verif/regress/benchmark.sh and all failed to run successfully 
some of the log files : 
1. [median.cv64a6_imafdc_sv39.log.iss.txt](https://github.com/user-attachments/files/15602638/median.cv64a6_imafdc_sv39.log.iss.txt)
2. 
[mm.cv64a6_imafdc_sv39.log.iss.txt](https://github.com/user-attachments/files/15602646/mm.cv64a6_imafdc_sv39.log.iss.txt)
and some more where all failed 

Can u please tell me what is wrong with my toolchain here 

And can u please comment on the procedure i have employed to running other C program in the previous message

JeanRochCoulon, 2024-06-06:
Are you on the up to date master branch, what is your hash ?

JeanRochCoulon, 2024-06-06:
If you clone the master branch (without any modification) and you simulate benchmark.sh, does it work ?

Tanishqgithub, 2024-06-06:
> Are you on the up to date master branch, what is your hash ?

Yes, I am on the master branch and here is the details of my hash :
[image]

JeanRochCoulon, 2024-06-06:
If you run your tests on origin/master, it should work

Tanishqgithub, 2024-06-06:
as you said i cloned the repository again and simulated the smoke test : 
[iss_regr_log.txt](https://github.com/user-attachments/files/15611083/iss_regr_log.txt)

then i simulated the benchmark.sh but the result i got is same as the previous message , none of the benchmarks are running successfully 
Please help me out in what is the mistake i may have been doing

Moschn, 2024-06-06:
I noticed that some of your logs have `cv64a6_imafdc_sv39` and your `run.py` uses `cv32a60x`. These are different configurations of the core, one in 64bit with floating point and MMU support and the other a 32-bit configuration. Make sure you use the correct compiler and the correct configuration you want to test. Also you could take a look at the CI which successfully executes the benchmarks and numerous more tests on every commit.

BTW: Your original question asked for `stdio.h`. CVA6 no longer supports riscv-pk which is required to print in simulation. Therefore, it will be tricky to print right out of the box.

JeanRochCoulon, 2024-06-10:
No update...    I suppose the @Moschn feedback solved the issue. I close it, re-open it if needed.

closed 2024-06-10 by JeanRochCoulon