openhwfoundation/cva6 #2202: [BUG][doc] incorrect set of pmpcfgN registers in CV32A65X annotated spec

zchamski, 2024-06-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The privileged spec annotated for CV32A65X (https://cva6.readthedocs.io/en/latest/06_cv32a65x_riscv/index.html) is misleading when describing `pmpcfgN` registers:

- The 16 PMP entries can be described by only four `pmpcfg` registers `pmpcfg0..pmpcfg3`.
- Figure 22 in section 3.7.1 should be replaced by an updated version that shows only four `pmpcfgN` registers.
- Since only 8 of the 16 entries are effectively implemented but all 16 are accessible (they do not raise `illegal instruction` exceptions on access), the behavior of the non-implemented entries needs to be specified ('read-only zero' for `pmpaddrN`? TBD for `pmpcfg2` and `pmpcfg3`?) See the discussion in issue #1178.

JeanRochCoulon, 2024-06-11:
#2232 fixed the RISC-V spec. @zchamski can you donfirm and close the issue if agree.

JeanRochCoulon, 2024-06-14:
Do you confirm this Github issue can be closed ?

zchamski, 2024-06-14:
Confirmed as FIXED, closing issue.

closed 2024-06-14 by zchamski