openhwfoundation/cva6 #2204: [Explanation please] <Struggling to understand what verilog is used in a given instance of CVA6>

ibooga, 2024-06-05:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

I have been reading the CVA6 code for a few weeks now and I'm struggling to understand what verilog files are being used when. In particular, my goal is to modify the cache system to test different cache designs. How can I tell which of the three cache designs are being used?

For example, when I run the hello world test, the ISA being used is "'rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei'". Trying to find what verilog files are used for this configuration has been very difficult for me. Please let me know if you have any suggestions.

Cheers

### Required Changes

A better description of what verilog files are being used in a given instance of CVA6.

### Current Status

Difficult to discern what verilog files are being used in a given instance of CVA6.

### Description of Done

A method for viewing the verilog files that are used in a given instance of CVA6.

JeanRochCoulon, 2024-06-06:
https://github.com/openhwgroup/cva6/blob/master/core/Flist.cva6 lists the used files.
If you woudl like to modify the configuration you are are simulating, the related config files are in https://github.com/openhwgroup/cva6/tree/master/core/include

ibooga, 2024-06-06:
Thank you, very much. I haven’t tried Flist.cva5, this is great news.I’ll give this a try and see how it goes. Thanks again!Cheers

JeanRochCoulon, 2024-06-06:
As the topic is solved, please close it and open another issue if needed

closed 2024-06-06 by ibooga