openhwfoundation/cva6 #2243: [BUG] SPIKE model is not configurable with TvalEn core option

LQUA, 2024-06-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

CV32A65X core has its TvalEn option set to 0.
The following tests are failed : 
- **rv32mi-ma_addr**
- **rv32i-I-MISALIGN_LDST-01**

The failure is due to the mtval value after the exception : the core set mtval to 0 (due to TvalEn) when SPIKE model is expecting the trapping address

In 64 bit configurations with TvalEn set to 0, **rv64mi-p-ma_addr** is also failed

This issue is linked to #1403

zchamski, 2024-06-14:
This is a duplicate of https://github.com/openhwgroup/core-v-verif/issues/2468.  Since the issue is on Spike side, it will be handled under https://github.com/openhwgroup/core-v-verif/.

valentinThomazic, 2024-08-26:
These tests have been disabled by https://github.com/openhwgroup/cva6/pull/2378 and should be enabled back on cv32a65x once this is fixed.

zchamski, 2025-01-09:
@valentinThomazic: The Spike configuration issue is solved using the general CSR write mask control and the appropriate changes in Spike Yaml config file (cf. https://github.com/openhwgroup/core-v-verif/pull/2490).   Please re-enable the tests and confirm that the behavior is now correct on CV32A65X.

@JeanRochCoulon: this **is** a CV32A65X issue ==> please adjust the label accordingly.

zchamski, 2025-01-23:
In the absence of @valentinThomazic I will re-test after reverting the relevant part of https://github.com/openhwgroup/cva6/pull/2378.

JeanRochCoulon, 2025-01-24:
As MTVAL is always zero in CV32A65X, this is not an issue for this configuration.
I remove the cv32A65x label.