openhwfoundation/cva6 #2244: assertion Cfg.NrPMPEntries <= 16 fails in  hello_world.cv64a6_mmu

AngelaGonzalezMarino, 2024-06-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

What is the reason for this assertion?  https://github.com/openhwgroup/cva6/blob/ade4c85e139eb4e2b743069e6d4f696d4a258e92/core/include/config_pkg.sv#L345 

MMU config pkg sets 64
https://github.com/openhwgroup/cva6/blob/ade4c85e139eb4e2b743069e6d4f696d4a258e92/core/include/cv64a6_mmu_config_pkg.sv#L67

So there is always a fail with simulators different from Verilator.

Should the assertion be removed/updated or is the MMU config setting an incorrect value?

JeanRochCoulon, 2024-06-12:
For information, 16 PMP regions are implemented in RTL, 64 is not possible

AngelaGonzalezMarino, 2024-06-12:
Thanks @JeanRochCoulon, then this would be an error in the MMU config?

@slgth what was the reason for this 64?

slgth, 2024-06-12:
This was defined to match our specific needs.
We can set it back to 16 for now, I'll do the PR.

Shouldn't we always test this parameter, regardless of which simulator is in use?

Thanks @AngelaGonzalezMarino!

Moschn, 2024-06-12:
@slgth adding support for 64 PMP entries in the RTL would not be too hard I think. Tell me if you actually need this, then I can quickly draft a PR to increase the maximum to 64.

closed 2024-06-14 by JeanRochCoulon

Moschn, 2024-06-20:
@slgth sorry took a while but the PR to increase the number of PMP entries to 64 is here: #2279

commit e3943f59 (2024-06-14): cv64a6_mmu: set NrPMPEntries to 16 (fixes #2244) (#2248)