openhwfoundation/cva6 #2245: Run make vcs to start the verdi waveform view some problems

cllll402, 2024-06-12:
I run make vcs in the cva6 root directory to open verdi and view the waveform.
[image]
[image]

However, verdi has the following error. And when I try to use the command run 1000ns, I find that the waveform is incomplete, and only 110ns waveform is displayed. Why? Please help me.
[image]
[image]
[image]

So I looked up the historical questions in the forum. Try to solve the problem. Follow #570  to the last step, but it seems to be the same problem: an invalid address for tohost. The trace log runs endlessly until it times out. So what am I supposed to do?
[image]

cllll402, 2024-06-12:
Considering that the DPI library was not recognized, I modified the command in the makefile to call it unconditionally.
[image]

JeanRochCoulon, 2024-06-12:
"make vcs" is not supported, please read the README.md to run a simulation with cva6.py
export VERDI=1 allows to activate the gui.
The issue can be closed

cllll402, 2024-06-12:
Okay, thank you.

closed 2024-06-12 by cllll402