openhwfoundation/cva6 #2247: latch detection in spyglass audit report

Asmaa-Kassimi, 2024-06-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Spyglass detects the presence of 33 latches in csr_regfile.sv. Refer to Audit.rpt report generated by design_audit command.

 Make sure that every signal in a combinational block is assigned in all possible conditions, to prevent from unintentional latches and further infer-latch  lint error in Spyglass.

JeanRochCoulon, 2024-06-12:
Thanks @Asmaa-Kassimi for this finding. I know you try to fix it, what is fine. If needed, we can provide support.

JeanRochCoulon, 2025-01-24:
@Gchauvon @yanicasa Several lint errors/warnings have been fixed, does this issue is fixed? Can we close?

Gchauvon, 2025-01-24:
It is fixed with #2727

closed 2025-01-24 by Gchauvon