openhwfoundation/cva6 #2268: [BUG] Design doc generation: custom CSRs are missing in the CV32A65X documentation

zchamski, 2024-06-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The riscv-config specification of CV32A65X contains the description of CSRs `ICACHE` and `DCACHE`, but these registers are not rendered in the design documentation.

The corresponding `riscv-config` output is located in file [config/riscv-config/cv32a65x/generated/custom_gen.yaml](https://github.com/openhwgroup/cva6/blob/master/config/riscv-config/cv32a65x/generated/custom_gen.yaml).

JeanRochCoulon, 2024-06-18:
@AbdessamiiOukalrazqou, I suppose this task could be completed by you ?   I assign it to you, feel free to re-assign if needed

AbdessamiiOukalrazqou, 2024-06-18:
@JeanRochCoulon ,@zchamski  it is under treatment

closed 2024-07-01 by JeanRochCoulon