openhwfoundation/cva6 #2280: [BUG] B extension: incorrect decoding for some instructions in RV32

ASintzoff, 2024-06-20:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

According to RISC-V ISA specification, for RV32, the bit 25 of instructions `BCLRI`, `BINVI`, `BSETI`, `BEXTI` and `RORI` must be equal to zero.
So when this bit is set, an illegal instruction exception has to be raised.

With the current implementation, when bit 25 is set, there is no exception for RV32.

```
end else if (instr.instr[31:26] == 6'b010010) instruction_o.op = ariane_pkg::BCLRI;
                else if (instr.instr[31:26] == 6'b011010) instruction_o.op = ariane_pkg::BINVI;
                else if (instr.instr[31:26] == 6'b001010) instruction_o.op = ariane_pkg::BSETI;
                else illegal_instr_bm = 1'b1;
```

```
  else if (instr.instr[31:26] == 6'b010_010) instruction_o.op = ariane_pkg::BEXTI;
                else if (instr.instr[31:26] == 6'b011_000) instruction_o.op = ariane_pkg::RORI;
                else illegal_instr_bm = 1'b1;           
```

JeanRochCoulon, 2024-06-20:
I agree with the Github issue, but which specification tells us that an exception is raised when bit25 must be equal to zero ?

AyoubJalali, 2024-06-21:
> I agree with the Github issue, but which specification tells us that an exception is raised when bit25 is equal to zero ?

if the 25th bit isn't equal to zero these instruction are treated as RV64 instruction base on the RISCV bitmanip spec

zchamski, 2025-01-17:
This needs to be checked on the Spike side as well.

AyoubJalali, 2025-01-17:
Spike has the same issue as i remember

AyoubJalali, 2025-01-20:
Sorry @MikeOpenHWGroup, I should be accurate on the version of the spec I'm referring to.

I the sepc version is : 

The RISC-V Instruction Set
Manual Volume I
Unprivileged Architecture
Version 20240411
Chapter 28. "B" Extension for Bit Manipulation, Version 1.0.0

zchamski, 2025-01-23:
The Spike side is fixed in https://github.com/openhwgroup/core-v-verif/pull/2578.

JeanRochCoulon, 2025-01-23:
RTL and Spike have been fixed. Close the item.

closed 2025-01-23 by JeanRochCoulon