openhwfoundation/cva6 #2298: [BUG] Linking error: undefined reference to htif_t::load_payload

shreyas-kalikar, 2024-06-27:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'm encountering a linking error while trying to build the CVA6 project with Verilator. Specifically, the linker is unable to find the definition for the htif_t::load_payload function, which is likely part of the Host-Target Interface (HTIF) library.
The full error message is:

`/opt/rh/devtoolset-11/root/usr/libexec/gcc/x86_64-redhat-linux/11/ld: ariane_tb.o:(.rodata._ZTV19preload_aware_dtm_t[_ZTV19preload_aware_dtm_t]+0x70): undefined reference to `htif_t::load_payload(std::string const&, unsigned long*, unsigned long)'`

Steps Taken:

I have tried the following steps to resolve the issue:

1) Changed verilator version to latest (Verilator 5.026)
2) Cleaned the build environment and attempted a fresh rebuild.
3) Changed CONFIG_NAME="gcc-13.1.0-baremetal" to CONFIG_NAME="gcc-master-baremetal" to check for changes

However, none of these have been able to get me past this error. The full error message above was present in hello_world.cv64a6_mmu.log.iss.
Please let me know if any further information is required, I would be very grateful, thank you!

JeanRochCoulon, 2024-06-28:
CVA6.py verifies the different tool versions before simulating.
What seems strange to me is the fact that you can change the Verilator version...
If checkers are disabled in your environment, this might explain the problem. I suggest to use the recommended tool version for Spike, Verilator and gcc.

shreyas-kalikar, 2024-06-28:
Sorry I did not explain it better, I went into cva6.py and changed the line that did the version check for verilator from 5.008 to 5.026 (However, this was just to explore an avenue/possibility). Also, by recommended tool version for gcc, would the default 13.1.0 count as recommended gcc version? Thank you!

JeanRochCoulon, 2024-06-29:
The htif function come from Spike. I would suggest you to start from scratch to re-install Verilator and Spike from scratch. Let's see if it helps.