openhwfoundation/cva6 #2317: [TASK] Upgrade Spike tandem with DPI functions to access CSRs and to update with Interrupt/Debug signals

AnouarZajni, 2024-07-03:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

To align RTL and Spike-Tandem in some test scenarios (example CSR tests, interrupt tests, ...)  and for functional verification purposes, upgrade Spike tandem is required to implement DPI functions to access internal spike CSR registers in read and write modes. Also, Spike should be aware about external interrupt/debug signals.

### Required Changes

Required DPI functions are:
1- DPI function to write spike internal CSR registers
2- DPI function to read spike internal CSR registers
3- DPI function to update spike with interrupt asserted/de-asserted 
4- DPI function to update spike with debug signals

### Current Status

N/A

### Risks

Some registers are not updated by spike so read return values will be wrong

### Description of Done

DPI function implemented and tested in UVM testbench

MarioOpenHWGroup, 2024-08-23:
#2407 The feature will be included when this PR is merged

zchamski, 2024-09-19:
For CV32A65X there are no debug features supported. This leaves the interrupt aspect open, see PR combo CVA6 #2502 + https://github.com/openhwgroup/core-v-verif/pull/2532 for a proposed solution.

closed 2024-10-26 by github-actions[bot]

zchamski, 2024-10-28:
Task completed with the merge of PR combo CVA6 https://github.com/openhwgroup/cva6/pull/2502 + https://github.com/openhwgroup/core-v-verif/pull/2532.  No debug feature needed as there is no debug support in CV32A65X.