openhwfoundation/cva6 #2321: [BUG] How to verify the output of an addi instruction?

ais77, 2024-07-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

What is the design heirarchy of the signals generated in the vcd file? for example, if i executed an addi instruction ,where can i find the signals for the output of the instruction, say in the alu unit?

JeanRochCoulon, 2024-07-04:
alu module is inside ex_stage module. The path can be extracted from your digital simulation tool.

dhanushmeda, 2024-07-05:
No when we run a C file we want to verify our output, how do we do that?

closed 2024-07-12 by Gchauvon

dhanushmeda, 2024-07-12:
The VCD files have hundreds of signals, how do we know which are the signals corresponding to the output signals of our C code.

Gchauvon, 2024-07-12:
You can simulate your C code with Spike our RISCV golden model and Verilator then compare both traces to make sure everything ran correctly.
If you want more detail on how is compiled your C code and how it will run in the CVA6, you should dump it to see the disassembly which contains a description instructions and registers used to create your C code.

JeanRochCoulon, 2024-07-12:
This is not the right place to ask for support! The github is close.
I suggest to use Github Discussions to ask for support.