openhwfoundation/cva6 #2328: Page not found for "build prerequisites" and "the toolchain itself" in README

Tanishqgithub, 2024-07-05:
I tried to access the link here in both : 
[image]

But both are not accessible , is it only form me or is some development going on in background 

1. For "build prerequisites" page in link [build prerequisites](https://github.com/openhwgroup/cva6/blob/master/util/gcc-toolchain-builder/README.md#Prerequisites) 
[image]

2. For "the toolchain itself" page in link [the toolchain itself](https://github.com/openhwgroup/cva6/blob/master/util/gcc-toolchain-builder/README.md#Getting-started)
[image]

valentinThomazic, 2024-07-05:
I think this is due to https://github.com/openhwgroup/cva6/pull/2320 @math-gout

JeanRochCoulon, 2024-07-06:
Fixed by #2332

closed 2024-07-06 by JeanRochCoulon