openhwfoundation/cva6 #2333: Error while building the toolchain

Syedateeb24, 2024-07-07:
After running the prerequisites command, while trying to build the toolchain i.e when running the following command 
"sh get-toolchain.sh gcc-13.1.0-baremetal" , i am receiving the following error :
[image]

Please help me fix this !

teobiton, 2024-07-07:
Have you tried running `bash get-toolchain.sh gcc-13.1.0-baremetal` instead of `sh` ?

G3voman, 2024-07-08:
It works, thanks

valentinThomazic, 2024-07-08:
@math-gout

G3voman, 2024-07-11:
Thu, 11 Jul 2024 17:07:22 INFO     GCC Version: 13.1.0
Thu, 11 Jul 2024 17:07:23 ERROR    Please clean up Spike by executing: rm -r tools/spike verif/core-v-verif/vendor/riscv/riscv-isa-sim/build
Thu, 11 Jul 2024 17:07:23 ERROR    You are currently using version - unknown - of Spike, should be: 1.1.1-dev 66cd091b. Please install or reinstall it with the installation script.

closed 2024-07-11 by JeanRochCoulon