openhwfoundation/cva6 #2334: [BUG] Addressing pitfall for beginners in Release page

akorb, 2024-07-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

[cva6](https://github.com/openhwgroup/cva6) builds as 64-bit by default, just as [cva6-sdk](https://github.com/openhwgroup/cva6-sdk). However, the [release page](https://github.com/openhwgroup/cva6/releases/tag/v5.0.1) of CVA6 contains a 32-bit build, which can only be understood by deciphering its file name `ariane_xilinx_cv32a60x.bit`, which is hard and very easy to oversee for starters in the CORE-V world. Contributing even more to the confusion is the release note **Fix cv64a6 configurations**, which makes it look like a 64-bit release.

As a result, it happens very easily that you use the CVA6 pre-built file (which is 32-bit), and build the CVA6-SDK with default arguments (which is 64-bit). And you then start to see weird stuff which is very hard to debug.

I think already more people ran into this problem. https://github.com/openhwgroup/cva6/issues/653 and https://github.com/openhwgroup/cva6/issues/1477#issuecomment-1742768679 (note for the latter that it works with Ariane 4.2, which maybe was released as 64-bit? Then it is clear why it worked with this version).

I propose two solutions:

- Put a clear warning text on the [Release page ](https://github.com/openhwgroup/cva6/releases/tag/v5.0.1)indicating that it is a 32-bit build
- Also release a 64-bit build, if there are two filenames which only differ in `32` vs `64`, it is much easier to understand. Otherwise you don't question it and might just assume a 64-bit build.

JeanRochCoulon, 2024-07-08:
@Gchauvon   I suppose you have an opinion on it ?

ASintzoff, 2024-07-12:
For the moment, a warning text has been added on the corresponding releases.

To provide a 64-bit build, some validation tests on such build should be done. As this is not the case today, 64-bit version is not yet part of the assets.

closed 2024-07-12 by akorb

isaar-ahmad, 2024-07-18:
Hi. I have a few related queries regarding 64-bit implementations of CVA6 :
> Is a 64-bit CVA6 implementation supported on Genesys-2 board? If yes, what is the configuration/target that should be selected ? 
> If no, is this planned for a future release?

Gchauvon, 2024-07-18:
Hi, it is supported. 
Configuration [cv64a6_imafdc_sv39](https://github.com/openhwgroup/cva6/blob/master/core/include/cv64a6_imafdc_sv39_config_pkg.sv) should be working. Others might work as well.

tbroadfoot, 2024-08-07:
Hello,

A follow up question: Is a new .mcs required for the ariane_xilinx_cv32a60x.bit version? In the previous 4.2 version a .bit and .mcs were provided, but now only a .bit is provided.

JeanRochCoulon, 2024-08-07:
The github issue is closed, please open another one if needed.