openhwfoundation/cva6 #2345: HPDCache is not compatible with WtDcacheWbufDepth > 2

cathales, 2024-07-10:
Hello César,

Setting WtDcacheWbufDepth to 4 instead of 2 would improve superscalar `cv32a65x` performance by 3%.

However, it seems to be incompatible with HPDCache as this configuration change makes smoke-gen tests fail, while using WT cache it works.

cfuguet, 2024-07-10:
> Hello César,
> 
> 
> 
> Setting WtDcacheWbufDepth to 4 instead of 2 would improve superscalar `cv32a65x` performance by 3%.
> 
> 
> 
> However, it seems to be incompatible with HPDCache as this configuration change makes smoke-gen tests fail, while using WT cache it works.

Hello @cathales,

There is no limitation on the number of entries (other than timing and area constraints) in the WB of the HPDcache so it should work.

How many bits do you have in the AXI interface ? You should have at least 3 for it to work: [000-011] are used for reads and writes and 111 is used for uncacheabke read and writes and AMOs

cfuguet, 2024-07-10:
And additional question, are you using the veto-testharness environment or the UVM environment for the smoke tests ?

cathales, 2024-07-10:
Good catch, UVM tests fail and smoke-gen runs using UVM. It might be a UVM issue and not a RTL issue.

@AyoubJalali what do you think about this?

AyoubJalali, 2024-07-11:
@AEzzejjari what do you think ?

JeanRochCoulon, 2024-07-22:
@AEzzejjari found and fix a bug in UVM #2379

@cathales you can try WtDcacheWbufDepth = 4 !

cathales, 2024-07-23:
Thank to @AEzzejjari for the fix! However it would cost almost 7 kGates for an CoreMark/MHz gain less than 3% so I would not do perform the change.

cathales, 2024-07-23:
Erratum: as this impacts only cache area, we will indeed increase this parameter.