openhwfoundation/cva6 #2350: [BUG] Incorrect depth of Instr_queue in the specification

AEzzejjari, 2024-07-10:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The instruction queue contains multiple instruction FIFOs to handle multiple parallel instructions at the same time. 
The number of FIFOs equals to INSTR_PER_FETCH, and the depth of each FIFO equals FIFO_DEPTH. 
The instruction queue also contains another FIFO for addresses in case of address prediction, with the same depth, FIFO_DEPTH. 
The actual depth of the instruction queue is not obvious due to this incompatibility between the address FIFO and the instruction FIFOs.

cathales, 2024-07-22:
In #2375 I change how the IQ is configured.

I have started working on superscalar documentation and will fix the IQ documentation there.

JeanRochCoulon, 2025-01-02:
@AEzzejjari We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it if it is the case ?

closed 2025-01-02 by AEzzejjari