openhwfoundation/cva6 #2351: [EXPLANATION OF SIMULATION LOG AND VCD] <Assistance understanding and mapping .log output to .vcd waveforms>

ibooga, 2024-07-11:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

I am running hello_world.c and trying to map the output .log file to the .vcd waveform.

Beginning with the hello_world.cv32a60x.log file, the first line of output is:

core   0: 0x0000000000010000 (0x00100413) li      s0, 1
3 0x0000000000010000 (0x00100413) x 8 0x00000001

Here's my understanding of these line pairs.

(Line 1)
(1A) "core 0: " The version of cva6 that I'm simulating has only one core, so every pair of lines begins with core 0
(1B) "0x0000000000010000" The program counter of the given instruction
(1C) "(0x00100413)" The Hexadecimal representation of the machine code instruction
(1D) "li      s0, 1" The assembly representation of the instruction
(Line 2)
(2A) "3" ???
(2B) "0x0000000000010000" Repetition of 1B, the program counter of the given instruction
(2C) "(0x00100413)" Repetition of 1C, the Hexadecimal representation of the machine code instruction
(2D) "x" ???
(2E) "8" ???
(2F) "0x00000001" Result of the instruction

Regarding the VCD waveform I have only been able to map, with certainty:
1C, hex of the current instruction, to ariane_testharness.i_ariane.i_cva6.orig_instr_id_issue[0][31:0]

Questions:
(Q1) Is there a resource explaining the structure of the .log output?
### I would greatly appreciate it if anyone could please explain what 2A, 2D, and 2E represent.
(Q2) What signals in the VCD output most quickly map to the .log output?
### For example, of the myriad pc signals, which one triggers the earliest and maps the most accurately to the instruction trace in the .log files.

###COMMANDS RAN###

DV_SIMULATORS=veri-testharness
cva6/verif/sim/cva6.py
--target
cv32a60x
--iss=veri-testharness
--iss_yaml=cva6.yaml
--c_tests
../tests/custom/hello_world/hello_world.c
--linker=../tests/custom/common/test.ld
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g ../tests/custom/common/syscalls.c ../tests/custom/common/crt.S -lgcc -I../tests/custom/env -I../tests/custom/common"

### Required Changes

See Task.

### Current Status

N/A

### Description of Done

See Task

Gchauvon, 2024-07-11:
Hello,
Regarding your first question : 

- 2A : execution mode when the instruction was executed, 3 = MACHINE MODE
- 2D,2E : hexadecimal representation of the writeback register : x8 = s0

The source of the log file is in "rvfi_tracer.sv". Most of the signals written in the log come from the commit instruction in the commit stage.

ibooga, 2024-07-12:
Thank you very much!

closed 2024-07-12 by ibooga