openhwfoundation/cva6 #2352: Non-Compiler based test environment

Tanishqgithub, 2024-07-11:
Respected Sir 

I am trying to take the necessary design files and the testbench files [sv files] to run on Xilinx Vivado , but couldn't map onto which are all the files typically needed.

I am requesting you help me in identifying necessary files for cva6 such that i have cva6_design.sv and the cva6_testbech.sv separately so that i can import these files in Vivado and execute these and analyze the waveform along with the power reports also. 

I think the design phase starts from this basic step itself i just need that files which are used to simulate the cva6 processor alone along with the testbench which provides the i/p Assembly level instructions [ISA] to the processor in each consecutive cycles.  

For example : 
Same as we did to simulate the RISCV single cycle processor in [https://github.com/Fede997/RISCV-PROCESSOR/blob/master/cpuRISCV-cpu_factorial.v](RISCV_SINGLE_CYCLE) : 
```
Here the factorial.dat is the file where the Assembly level instructions are given in hexadecimal format to RISCV_CPU
mentioned here : 
![image](https://github.com/openhwgroup/cva6/assets/108278729/92c633cd-24c9-4014-8c16-d9e2afed596e)

```

I know that the design phase starts with this type of file orientations so can u please share that files so that i can easily execute the sv files in vivado ?

Thanking You

Gchauvon, 2024-07-12:
Hello, 

Xilinx Vivado is not supported as a simulator for CVA6. We recommend you to use Verilator as a free opensource simulator. Steps to produce a waveform are explained in the README.

JeanRochCoulon, 2024-07-23:
To build the bitstream, use the "make fpga" command. I suggest to close the issue if ok for you.

closed 2024-07-24 by Tanishqgithub