openhwfoundation/cva6 #2365: Spyglass Warnings increased by new PMP entries

Asmaa-Kassimi, 2024-07-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Updating the number of PMP entries increased the number of Spyglass warnings, mainly W240 "An input has been declared but is not read".

In most cases, the input should be removed, but will force a change in design units instantiating this module, or waived, if the input appears as a place holder for later logic insertion.

JeanRochCoulon, 2024-07-24:
I agree !

JeanRochCoulon, 2025-01-02:
As this issue is related to Spyglass, I assign @ASintzoff. Waiver or not, this need to be discussed.

JeanRochCoulon, 2025-01-07:
As PMP is not enabled for 60x and 65x, I removed the CV32A65X label