openhwfoundation/cva6 #2366: [BUG] Bug Report: Spike Version Mismatch

sudorair, 2024-07-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I encountered an error related to Spike simulator version mismatch while running the cva6.py script.

Steps to Reproduce:

1. Cloned the CVA6 repository: git clone https://github.com/openhwgroup/cva6.git
2. Set up the environment and dependencies as per the README.
3. Attempted to run cva6.py with the following command:

source verif/sim/setup-env.sh
export DV_SIMULATORS=veri-testharness
cd verif/sim
python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"

Expected Behavior:
The script should execute successfully without errors related to Spike version.

Actual Behavior:
Encountered the following error:

[image]

Additional Information:

Tried removing and reinstalling Spike, but i am getting the following error while trying to run the following command 
_sudo bash verif/regress/install-spike.sh_

[image]

Environment details: I am running in Centos 7. Let me know if there is any other details you need.

Please advise on how to resolve this issue or if there are any specific steps needed to correctly set up Spike for the cva6.py script.

JeanRochCoulon, 2024-07-13:
I think you meed to make your env support cmake. @zchamski can you elaborate on it?

valentinThomazic, 2024-07-15:
A few questions that might help :

- Did you init the git submodules recursively ?
- Is cmake installed on your system ? Which version are you using ? Should be cmake 3.4 or higher.

By the way, you don't need to run the `install-spike` script as root.

sudorair, 2024-07-16:
Thanks for the response. Yes I tried intalling cmake and it worked. I was able to run the hello world.

closed 2024-07-18 by valentinThomazic