openhwfoundation/cva6 #2370: [BUG] Violation of PMA : Unauthorized  Access to Vacant Memory Region

abdelhak-chkirid, 2024-07-17:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Access to vacant regions (e.g., the address 0xc0000100) is allowed for read and write operations with no exceptions appearing.
Addresses within the vacant region should not be accessible. All read/write operations should  result in an exception.

JeanRochCoulon, 2024-07-22:
@abdelhak-chkirid   The riscv specification for 65x has been updated. Consider it: https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#pma 
[image]
We simplified the specification by not supporting PMA inside CVA6.
To me, this Github issue can be close.

JeanRochCoulon, 2024-07-23:
@abdelhak-chkirid   Can we close this issue ?

abdelhak-chkirid, 2024-07-24:
yes , of course

closed 2024-07-24 by abdelhak-chkirid