openhwfoundation/cva6 #2376: scoreboard.sv [BUG] : RTL assertion failed

AyoubJalali, 2024-07-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

For now I don't know the real cause of this issue, but I wanna file it in github.
An assertion is failed in CVA6 scoreboard.sv : 
[image]

to regenerate the problem :

cathales, 2024-07-22:
According to the simulation waveforms, there is an issue in the `load_unit`. A request is made with transaction id 6, then it returns with transaction id 7.

JeanRochCoulon, 2025-01-02:
@cathales We are cleaning the "Github Issues". What is the status of this issue ?

cathales, 2025-01-06:
I think @yanicasa is more aware about what happens in the `load_unit` than I am.

yanicasa, 2025-01-24:
On the future CV32A65X branch, with the OBI buses, the logic has been updated a lot, there is more checking on common exceptions from mmu/Missalign/pmp which go back to the load/store/amo to be sure not to catch an exception an exception intended for an other unit. 
Since these modifications we have not reproduced the problem, the test is stable on the pipelines

The label can be updated notCV32A65X

khandelwaltanuj, 2025-02-05:
Hello,

I just observed this assertion failing as well while running smoke-gen_test.sh for cv32a65x. What surprises me a little is even if assertion fails the test is not signaled as failed. It feels like as if only mismatched between spike and simulation are considered as failure ? 
Is it something that you have observed or I need to do may be more manipulation to get the exact result of the run ? 

Thanks and Regards
Tanuj

JeanRochCoulon, 2025-02-05:
I think this issue is already reported in #1624 
If it is the case, thanks to close this issue, otherwise please elaborate.

khandelwaltanuj, 2025-02-06:
Yes this is the case 

Thanks