openhwfoundation/cva6 #2382: [BUG] <title> Opening files in Vivado and sythesizing gives errors about missing ports and missing ethernet connections

walrider3, 2024-07-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Ok so basically, I need to use CVA6 to add a peripheral to it. For that I want to open the design in Vivado. However, when I add the files in the cva6/core and cva6/core-apu files, I get errors like there are some configuration inputs missing e.g. SuperscalerEn, TechnoCut etc. I set those values myself then accordingly, but after that the ethernet connections are left.
e.g. [Synth 8-1031] eth_txck is not declared ["C:/Users/PMLS/Downloads/cva6/corev_apu/fpga/src/ariane_xilinx.sv":899]
Please help

JeanRochCoulon, 2024-07-22:
SuperscalarEn and TechnoCut have been freshly added to cva6. Can you double check the coherency between cva6 and cva6/core-apu files?

walrider3, 2024-07-23:
> SuperscalarEn and TechnoCut have been freshly added to cva6. Can you double check the coherency between cva6 and cva6/core-apu files?

What do you mean by checking the coherency between the 2 files?
Also what can I do about the ethernet connections?

JeanRochCoulon, 2024-07-23:
Can we decorrelate the two problems...
Without adding the ethernet, it should work without discrepancy of SuperscalarEn. Can yo confirm it?

walrider3, 2024-07-23:
I dont get what you mean by decorrelation, I mean the two problems are seperate problems they are not interconnected.
Also, I haven't added any ethernet, its just that in the top level module of ariane_xilinx.sv, the eth_xxxx inputs aren't given any inputs, so they are saying its a parameter needs to be passed:
 i_ariane_peripherals (
    .clk_i        ( clk                          ),
    .clk_200MHz_i ( clk_200MHz_ref               ),
    .rst_ni       ( ndmreset_n                   ),
    .plic         ( master[ariane_soc::PLIC]     ),
    .uart         ( master[ariane_soc::UART]     ),
    .spi          ( master[ariane_soc::SPI]      ),
    .gpio         ( master[ariane_soc::GPIO]     ),
    .eth_clk_i    ( eth_clk                      ),
    .ethernet     ( master[ariane_soc::Ethernet] ),
    .timer        ( master[ariane_soc::Timer]    ),
    .irq_o        ( irq                          ),
    .rx_i         ( rx                           ),
    .tx_o         ( tx                           ),
    .eth_txck,
    .eth_rxck,
    .eth_rxctl,
    .eth_rxd,
    .eth_rst_n,
    .eth_txctl,
    .eth_txd,
    .eth_mdio,
    .eth_mdc,
    .phy_tx_clk_i   ( phy_tx_clk                  ),
    .sd_clk_i       ( sd_clk_sys                  ),
    .spi_clk_o      ( spi_clk_o                   ),

Moschn, 2024-07-23:
Just adding the files will not work, as you will have to define some parameters and extra files/IP. We do not currently have a readme how to create a vivado project manually. If you want to see what is happening in our vivado build, take a look at the `make fpga` command and what it does behind the scenes.

JeanRochCoulon, 2024-07-23:
Thanks @Moschn

walrider3, 2024-07-24:
I basically need to connect a peripheral (DLA as in deep learning accelerator), so I was thinking I would have to open the design the block design in Vivado and make that connection. It could also be possible to add it in code, but I need the header file(.h) that contains the memory mapping for the peripherals. I found the address map in the top level module ariane_xilinx.sv and it corresponding peripheral folder where the peripherals e.g. UART is defined. I just need to know where their addresses are defined so I can add my own peripheral's address.

walrider3, 2024-07-25:
Also, when booting linux on this using the prebuilt image. Do I have to just flash the bbl.bin file into the sd card? Is that the only file required? I followed the following instructions: https://github.com/PrincetonUniversity/openpiton?tab=readme-ov-file#booting-smp-linux-on-genesys2-or-vc707

Moschn, 2024-07-25:
Openpiton instructions are different and I do not know if they apply to our FPGA image. I believe the instructions in CVA6-sdk should work. 

If you want to use openpiton then you should also use their FPGA image.

walrider3, 2024-07-27:
> crossbar

Do you mean a crossbar switch as described here?https://www.tutorialspoint.com/what-is-crossbar-switch-in-computer-architecture

wal222, 2024-07-28:
> Our setup is not done in the Vivado block design. I also don't think it is easy to convert our design/IPs to Vivados block design. So I believe you probably have to do it in code.
> 
> What do you mean by header file (.h)? The core is designed in systemverilog and IPs are connected to the bus via a crossbar. This is not the same as a C header. I assume you will have to attach your peripheral to the crossbar (e.g., [here for the peripheral memory map](https://github.com/openhwgroup/cva6/blob/14be0af7f017eef8fec615308c1d161584b45f07/corev_apu/fpga/src/ariane_xilinx.sv#L326) and[ here for the addresses](https://github.com/openhwgroup/cva6/blob/14be0af7f017eef8fec615308c1d161584b45f07/corev_apu/tb/ariane_soc_pkg.sv#L57)). There is also the option to use [CVXIF](https://github.com/openhwgroup/core-v-xif) but you will have to ask other people who have experience with this interface. After the peripheral is connected, you should be able to access it at the memory address that you configured in the crossbar.

Could you give any references or resources as to how I would go about connecting my peripheral to the crossbar? Also, what would I have to change in the software stack then. Would I have to change the kernel image, or the device tree? The cva6 sdk repo has only bbl.bin file and a bbl file(type not specified). I don't know what the bbl file is. Both bbl and bbl.bin are the same size of 14.2 Mb.

Moschn, 2024-07-29:
A crossbar is just a simplified interconnect where every master is connected to every slave. 

You will need some basic knowledge of system-verilog to connect things to the interconnect. I do not know of resources to quickly learn how to do that. 

You do not just have to add it to the memory map. Take a look at where these memory addresses are used. What you mention is just the definitions of the addresses, but they are instantiated somewhere else. 

I do not remember exactly how interrupts are routed. I believe the risc-v spec specifies a single interrupt signal per core, with an extra interrupt controller. There are a bunch of interrupt controllers in the risc-v space: CLIC, CLINT, PLIC, etc. I assume CVA6 uses one of these. Also, I believe that interrupts are not routed through the interconnect. 

The software stack that you use seems weird. BBL used to be the old bootloader, but I think we now swaped to openSBI. So I assume you are still using the openpithon flow? I really would recommend you to use the CVA-sdk flow. Nevertheless, the bbl.bin file is a binary file containing the bootloader and the kernel. 

After you connect your peripheral to the core, you can just access the physical address of the peripheral from any software. The device tree is usually only used for drivers to understand what devices are present and at which address. In your case, I guess there exists no driver yet in the kernel. Therefore, you probably also do not have to add it to the device tree.

walrider3, 2024-07-29:
> A crossbar is just a simplified interconnect where every master is connected to every slave.
>
> You will need some basic knowledge of system-verilog to connect things to the interconnect. I do not [...]

I am following the CVA6-sdk flow, it said to use the prebuilt images in the link they provided. In that link, there was only a bbl.bin file and bbl file both of the same size. So I flashed the bbl.bin file and got the following output:

Moschn, 2024-07-29:
Oh the release is extremely old. I suggest trying to build an image yourself and flashing the sd card with the command in the readme

walrider3, 2024-07-29:
Running the "make images" command doesn't work. I first tried it in my own environment. Then I tried in the docker container. I got the following errors:
[image]
I checked sources.buildroot.net/linux/linux-v5.10.7-br1.tar.gz, and it actaully doesn't exist. Then looked into the sources.buildroot.net/linux directory and it doesn't exist in there either, so I tried wget-ting the version [linux-5.7.10.tar.xz] but that didn't work either since it said the destination directory doesn't exist. I haven't been able to solve this for past week or so.